nordicsemi/IOS-nRF-Mesh-Library #11: Checking out the Mesh

gabdsg, 2018-05-03:


gabdsg, 2018-05-03:
Now on proxy_client i'm getting this error: Ignored. Node is not configured.
I checked client_publication_configured() and (pub_addr_handle == DSM_HANDLE_INVALID) is true for every model

mostafaberg, 2018-05-03:
First of all congratulations on the first issue :)

Secondly, which branch are you working on ? I recommend building from develop as the current initial release is a bit incomplete and there are lots of fixes on dev.

There’s not so much to do besides flashing the sample firmwares, all you need to do is just hit the add button and the provisioning and configuration will take place straight away , then you’ll be able to see the newly provisioned node on the main network view where you can configure it further and view all its information. 

I’ll get back to you again tomorrow morning to see how far you went and if you have any more issues :)

Thanks for the early testing !

mostafaberg, 2018-05-09:
Do you still have any issues with the library or can we close this now ?

gabdsg, 2018-05-09:
it works! Thanks

mostafaberg, 2018-05-09:
That's great news ! glad to hear you sorted things out !
please let us know if there are any new issues or feature requests :)

closed 2018-05-09 by mostafaberg