nordicsemi/IOS-nRF-Mesh-Library #17: Swift Compiler Error: Expression was too complex to be solved in reasonable time; consider breaking up the expression into distinct sub-expressions

amigomcu, 2018-05-14:
**Describe the bug**

Expression was too complex to be solved in reasonable time; consider breaking up the expression into distinct sub-expressions

 let packedNetKey = Data([netKeyIndex[0] << 4 | ((netKeyIndex[1] & 0xF0) >> 4),netKeyIndex[1] << 4])

    private func discoveryCompleted() {
        if let provisioningData = provisioningData, meshState != nil {
            logEventWithMessage("provisioning started")
            let meshStateObject = meshState.state()
            let netKeyIndex = meshStateObject.keyIndex
            let packedNetKey = Data([netKeyIndex[0] << 4 | ((netKeyIndex[1] & 0xF0) >> 4),netKeyIndex[1] << 4])

**To Reproduce**
clone IOS-nRF-Mesh-Library and open xcode project
(nRFMeshProvision.xcworkspace)

build project , then it show "Expression was too complex to be solved in reasonable time"

**Expected behavior**
 it too complex for swift , that should be change multi statment assign packedNetKey

**Platform details:**
 -Xcode 9.2
 -Swift 4
 - Device: iPhone6
 - OS:  iOS 10.3.2
 - Library Version (or Commit)
 
**Logs / Screenshots**
Add any logs or screenshots that can help us debug the issue faster

http://220.134.203.121/nextcloud/s/MBAXtxFjRPX82GC

mostafaberg, 2018-05-14:
Hi @amigomcu !, sorry for the inconvenience,

We've seen issues like that on some computers, i want to confirm that the solution will work for you before committing a change, what i'll do is to split up the bit manipulations into multiple steps instead of one.

Can you try to change the line:

`let packedNetKey = Data([netKeyIndex[0] << 4 | ((netKeyIndex[1] & 0xF0) >> 4),netKeyIndex[1] << 4])`

do this:
```
var tmpOctet1 = netKeyIndex[0] << 4
var tmpOctet2 =  netKeyIndex[1] & 0xF0
tmpOctet2 = cmpOctet2 >> 4
let firstOctet = tmpOctet1 | tmpOctet2
let secondOctet = netKeyIndex[1] << 4
var packedNetKey = Data([firstOctet, secondOctet])
```

If this resolves your issue i'll have a proper bug fix in the next version so you won't have to modify the code yourself :)

amigomcu, 2018-05-15:
hi mostafaberg 
it is work now ,
Have any document how to use to build mesh network ? 

Provisioning,Configuration,Composition data,Application keys and Publishing and subscribing

thank.

closed 2018-05-15 by amigomcu

mostafaberg, 2018-05-16:
@amigomcu Good to hear it works, will include that in the next update !

The documentation will come as soon as we have nailed down all the features, for now you can open tickets on devzone with your questions and we will guide you through it quickly, sorry about the delays with the documentation:)