nordicsemi/IOS-nRF-Mesh-Library #40: Can copy an appkey and add it into nRF Mesh App

kaiser-ren, 2018-06-23:
Currently, nRF Mesh allows to add an appkey, but this key is generated randomly. 
Whether can add a feature that user can copy and paste an appkey by themselves, this will make nRF mesh can work with other provisioners together like BlueZ.

mostafaberg, 2018-06-25:
Good morning @kaiser-ren !, this is a feature we are going to implement soon yes :), most probably this will come in the next update, you'll be notified when it's implemented when I close this issue :)

trusbe, 2018-06-26:
We may use a QR code to share keys.

mostafaberg, 2018-06-26:
@trusbe what keys do you share?, QR is being considered, but not planned and might not make it (as you can easily use a 3rd party app to scan codes and paste them, so probably first iteration will allow you to paste values into nRF Mesh

trusbe, 2018-06-27:
@mostafaberg We are now planning to share the locally saved MeshState, which is actually shared with NetworkKey and AppKey. The shared data is encrypted. 

As a result, the network can be shared to another mobile phone, and one mobile phone can manage multiple networks.

mostafaberg, 2018-06-28:
Perfect, we are actually going to add the export/import feature followign the Bluetooth database spec so it should be interoperable between any mesh client.

Will keep you updated about the QR decision, but for now it's still just in talks and an early stage.

mostafaberg, 2018-07-19:
@trusbe just saw your last question, to know which devices are in a group address you'll have to parse the state of the mesh and see all nodes that are either publishing of subscribing to that address.

trusbe, 2018-07-26:
@mostafaberg  Understand, I saw that some of the Messages do not have a TID set. Is the location of this parameter the first in the payload? When can I not need it?
 e.g:

NO TID
```
    public init() {
        opcode = Data([0x82, 0x01])
        payload = Data()
    }

There is a TID but the position is not the first
```

```
    public init(withTargetState aTargetState: Data) {
        opcode = Data([0x82, 0x02])
        payload = aTargetState
        //Sequence number used as TID
        let tid = Data([SequenceNumber().sequenceData().last!])
        payload.append(tid)
    }
```

mostafaberg, 2018-10-08:
Key input, copy/paste too has been implemented and works side by side with the generate method

closed 2018-10-08 by mostafaberg