nordicsemi/IOS-nRF-Mesh-Library #41: Support CocoaPods ?

trusbe, 2018-06-25:
This can be synchronized directly to the latest update.

mostafaberg, 2018-06-25:
The library is currently working as a private pod due to some difficulties with OpenSSL.
You may use it as a pod internally, but you'll need to modify the OpenSSL dependency and remove it form our Pod and add it to your dependency list, we will have a fix for that eventually and release as a public pod, but for now this is not fully supported.

Also, this is a duplicate of #23 so I'll close this for now and you can follow that issue instead.

as soon as we fix that, you'll be notified when the issue is resolved and closed so I'd suggest you follow the issue #23 instead.

thanks a lot for your feedback.

closed 2018-06-25 by mostafaberg

trusbe, 2018-06-25:
Thanks for your feedback, 

Oh, I got it, Waiting for good news.