nordicsemi/IOS-nRF-Mesh-Library #43: Can I increase the switching operation and status return for all devices?

trusbe, 2018-07-26:
Use group address 0xFFFF for all device switches and get the status of each device in the delegate.

mostafaberg, 2018-07-26:
Problem is you can't subscribe to `0xFFFF` as it's a special address (some firmwares will accept that, but I don't think it'll work as you intend it to)

You are attempting to get updates from Generic On Off servers so you can update the state in the UI ? or are you doing something else ?

mostafaberg, 2018-07-26:
`0xFFFF` is a fixed address, if you see Mesh Core spec document V1.0 on page 42, you'll see the types of fixed addresses are:

`0xFFFC` => All Proxies
`0xFFFD` => All Friends
`0xFFFE` => All Relays
`0xFFFF` => All nodes

Messages with `0xFFFF` destinations, will only be handled by the primary elements of all nodes in the network. so you are already "Subscribed" to that on the nodes.

Generic OnOff for example will not publish to `0xFFFF` when their state changes, you have to actually subscribe your phone to the same group address for every server, so when a switch sends a message, you'll get the state notification, and then you'll need to manually poll the state form the individual nodes.

You can do that using the `GenericOnOFFGet` Message, found in Mesh Model specification V1.0 on page 42.

There might be other ways, but I'm a bit uncertain, I'm also not sure if the generic models will send out a publication when the state changes, so even if you have a publish address on the generic on off server, I don't know if that will actually publish or not, have you tried that ?

philips77, 2019-08-20:
Closing the issue as the old version is no longer maintained.

closed 2019-08-20 by philips77