nordicsemi/IOS-nRF-Mesh-Library #58: Implement DFU feature

mostafaberg, 2018-07-27:
As a *power user*, I want to be able to flash mesh nodes with newer firmwares so that I can easily test, update and verify network operation with the latest firmware versions.

trusbe, 2018-07-31:
+1，Urgently need this feature to upgrade the entire network.

mostafaberg, 2018-07-31:
We haven't spec'd this yet, but we are planning to do so this coming week, will keep you updated about the roadmap:)

trusbe, 2018-08-01:
Nice!  👍👍👍

BubblyNetDev, 2019-10-02:
Any future plans to add DFU to the sample app?

philips77, 2019-10-03:
Not in the coming months. I have to switch to another project for some time.

BubblyNetDev, 2019-10-03:
Okay, when are you moving on? What will be the state of the application?

philips77, 2019-10-03:
I'll transition after I'll release the version 2. I'll be also fixing bugs after that.
The version 2 is almost ready, there are some more features I want to add, e.g. database migration from version 1.

olsky, 2019-10-30:
+1 very important, a must have :)

philips77, 2019-10-30:
We understand, but I'm really busy with other things :( 
And the official spec for mesh dfu isn't even released yet!

PM-BubblyNet, 2019-12-15:
Hey @philips77 about the "And the official spec for mesh dfu isn't even released yet!" we also notice that it has being pushed back, originally expected on spring 2020 now its summer 2020... so got me thinking if maybe there is a case about implementing BLE DFU as a temporary solution? I wonder how difficult from the app and firmware would this be? is it temporary solution worth exploring? or the complexity for just few months is not really worth it?

philips77, 2019-12-15:
I didn't read the draft, so I don't know how much work would it require. We also don't have it out backlog now.

PM-BubblyNet, 2019-12-15:
I see... we checked the current DFU from others like Cypress and Silvair, they don't seem to be a "Mesh" DFU, instead they seem to update 1 device at the time using several (4) proxies as a "queue", my idea was to make something like that as a temporary solution.
However this brings few questions, for example all the nodes needs to have proxy enabled, we actually disable most of the proxies, then someone needs to go walking very slowly trough the whole site updating 1 device at the time... all in all this "mesh" BLE DFU doesn't seem a very good idea, but still maybe worth it as a temporary solution until the real Mesh DFU model comes from the SIG?

trusbe, 2021-07-20:
I would like to ask a few questions about DFU:

1. Is there a DFU plan that supports the same device firmware in the Mesh network?

2. Understand that all current new functions only support NCS SDK, nRF5 SDK no longer updates new functions, only fix bugs, currently we are using firmware written by NCS SDK, chip memory must be 512K or more to support separate DFU, Is there a better way to support DFU for chip devices with 512K or less memory?

Thanks.

philips77, 2021-07-22:
> Is there a DFU plan that supports the same device firmware in the Mesh network?

I'm afraid I don't understand.

> Understand that all current new functions only support NCS SDK, nRF5 SDK no longer updates new functions, only fix bugs, currently we are using firmware written by NCS SDK, chip memory must be 512K [...]

Use an external flash memory to download the new firmware, MCUboot which is used for firmware update in NCS SDK can be used with external flash. For more information, please check [infocenter](https://infocenter.nordicsemi.com) or [DevZone](https://devzone.nordicsemi.com).

trusbe, 2021-07-22:
> Is there a DFU plan that supports the same device firmware in the Mesh network?

It means that only one device needs to be upgraded, and the Mesh network will automatically upgrade other devices of the same device type. For example, if there are five lights in the same network, one of the devices will be upgraded, and the other four will be upgraded automatically.

Thank you very much, it seems that the only way to increase external storage or use other chip models.

philips77, 2021-07-23:
So there will be many ways you may do DFU. The best one from mobile phone's point of view is indeed initiating DFU using mesh protocol, then sending a new firmware using a OOB (Out Of Band, not through mesh proxy service, but something much faster) and confirming with mesh protocol. Then the proxy node (with dfu capabilities) would be responsible for transferring it to all nodes registered during the initialization phase. You send once in 1 minute, and the image distribution takes very long time.

smenor, 2022-04-19:
Is this ever coming ?

philips77, 2022-04-20:
As far as I know, the official specification is still under development. Realistically speaking, I don't see it coming to mobile implementations in 2022.

smenor, 2022-04-20:
Thanks‥ unfortunate but good to know

Is there a good example alternative using mesh and DFU over BLE in the meantime or should we just hold off on BLE mesh implementation ?

olsky, 2023-10-16:
it has been released: https://www.bluetooth.com/mesh-device-firmware-update/

;-) any hints @philips77 ?

philips77, 2023-10-18:
Yes. With 4.0 released the DFU will come in 4.1. I need to switch to Android for a bit to fix some issues here and there and I'll start updating all my mesh nodes like a crazy person. Hopefully by end of this year you'll be able to join me and we will update the world!

olsky, 2023-10-18:
Great news! I will join you, the world will be updated by two crazy ones :)

philips77, 2023-10-18:
On Android DFU will come later. We're rewriting the lib in Kotlin (should make it public as soon as it's "usable enough") and we need to focus on implementing the basics first.

olsky, 2023-10-18:
When basics first, then the initiator role is good starting point, right?

philips77, 2023-10-18:
For DFU - yes. For transport of the image we will use SMP protocol and [nRF Connect Device Manager library](https://github.com/NordicSemiconductor/IOS-nRF-Connect-Device-Manager/), sending meta-data over mesh. Then the DFU proxy should take over and slowly forward the image to desired nodes. That way we'll get 30 kB/s from the phone instead of 0.01.

philips77, 2023-10-18:
I mean, in nRF Mesh app. Library will just have API to do whatever.

olsky, 2023-10-18:
thinking of speed, sure: faster is better :)

I think if the fw is moved to distributor, then the speed is not that relevand (distributor will push updates and network will be updated). It would be nice to see in the Lib the minimum required for the Initiator Role: 
- Firmware Distribution Client
- Firmware Update Client
- BLOB Transfer Client
then UI/Views can catchup later...

would that fit into our roadmap?

philips77, 2023-10-18:
I didn't memorize the whole DFU spec yet, like I did the other parts of mesh.

We will use mesh models for sending metadata, but will not use BLOB transfer to send the image to Distributor. Instead we'll use OOB method - McuManager. However, I'll try to implement the upload in transport agnostic way so users can choose other protocols. And, depending on complexity, perhaps we can add BLOB as well, just for completeness. As I said, this amazing novel is waiting for me.

olsky, 2023-10-18:
Oh, that's a great plan! A sound-crazy-plan thing 🎢

olsky, 2024-02-07:
Hi, just a quick follow-up: any updates on this?

philips77, 2024-02-08:
Hi, it will come in few months. I'm busy with some other tasks at the moment, but DFU comes next.
Also, as we're working on a new Kotlin mesh library for Android, we want the new one to catch up with the is one in feature set, so we can start adding new features in parallel.

olsky, 2024-10-23:
hey @philips77, hope all good at your side!
could you please share update on the topic? )

philips77, 2024-10-23:
Hi, I'm starting to implement DFU very, very soon. Need to close some old topics, but this or early next week.

On iOS, that is. On Android side we need to finish rewriting the current lib into Kotlin (almost complete) before adding new features, and there are some that will come before DFU, like Remote Provisioning.

olsky, 2025-06-02:
Hi @philips77 , just in case you need a dfu tester ;-)