nordicsemi/IOS-nRF-Mesh-Library #61: iOS 9 support

mostafaberg, 2018-07-27:
As an *iOS 9 user*, I want to be able to run nRF Mesh on my device so that I can use mesh features on my device.

philips77, 2019-07-16:
The new lib is available for iOS 9+. Only the sample app requires 10+. At least i hope so.

huangdaxiaEX, 2019-07-19:
@philips77  Hi, thank you for your reply.

 I found the new lib source code use iOS 10+ API when received public key data [SecKeyCopyKeyExchangeResult](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/7cd48f4aeb04147cfb3aa1926c64afb65b3342f1/nRFMeshProvision/Classes/Provisioning/ProvisioningData.swift)

That can't work on iOS 9+.

philips77, 2019-07-19:
Oh, great that you found this. Do you know any alternative we could use instead?

philips77, 2019-07-19:
I guess we could try the OpenSSL lib...

huangdaxiaEX, 2019-07-22:
Yes 👍 

OpenSSL lib provided it, I will try it.   :）

BTW,  the `SecKeyCopyKeyExchangeResult` works on iOS 13 will  receive an error "OSStatus-50- kSecKeyKeyExchangeParameterRequestedSize is missing". 

Need described `kSecKeyKeyExchangeParameterRequestedSize` length.

[image]

philips77, 2019-07-22:
Again, nice catch! Thanks!

huangdaxiaEX, 2019-07-23:
@philips77 

I seem to have fixed it.

use 
`kSecKeyAlgorithmECDHKeyExchangeStandard` 
instead 
`kSecKeyAlgorithmECDHKeyExchangeStandardX963SHA256`

and

`exchangeResultParams` describe `kSecKeyKeyExchangeParameterRequestedSize : 32` 

The iOS 13+ will work fine.

philips77, 2019-08-01:
Hi, could you check if it works on iOS 13 now with the latest changes?

huangdaxiaEX, 2019-08-13:
I use iOS 13 beta.6, it works fine, thanks!
👍

harendra23, 2019-09-12:
Hi,
I'm facing the same issue. regardless you pass parameter requestSize or not, iOS13 always has missing kSecKeyKeyExchangeParameterRequestedSize issue.
I tried on iOS beta 7, issue persist.
Could you please help me how to fix this issue?

Thanks,

philips77, 2019-09-12:
Checking now.

TuYuan, 2020-06-10:
Hi，philips 
     As an iOS 9 user, I want to be able to run nRF Mesh on my device so that I can use mesh features on my device. What should I do？

philips77, 2020-06-10:
Hello,
For now, supporting iOS 9 is not our priority. This task is here for us to remember about this. As far as I remember there are just few features that are not supported on 9, and mainly in the sample app. But I may be wrong here.

TuYuan, 2020-06-11:
Hi，philips
Can the new lib source code support iOS9? If supported, give a link. Thank you!

philips77, 2023-06-16:
As iOS 9 is now many years old I'm closing this never-completed task.

closed 2023-06-16 by philips77

mostafaberg, 2023-06-16:
About time @philips77 ?😂