nordicsemi/IOS-nRF-Mesh-Library #73: nRF Mesh app stuck during provisioning of light switch proxy client, mesh sdk v2.1.1

bkvaale, 2018-07-31:
**Describe the bug**
The nRF Mesh app is stuck at 91% in the progress bar. The full nRF Mesh log is attached below:

[image]
[image]
[image]
[image]

The log from the light switch proxy client is attached here:

<t:          0>, main.c,  331, ----- BLE Mesh Light Switch Client Demo -----
<t:        538>, main.c,  242, Initializing and adding models
<t:       5449>, main.c,  374, Device UUID : 0059ABCDEFABCDEFACCDEFABCDEFABCD
<t:     456106>, main.c,  264, Successfully updated connection parameters
<t:       1030>, main.c,  101, Successfully provisioned
<t:       1038>, main.c,  109, Node Address: 0x0001 
<t:      79085>, main.c,  264, Successfully updated connection parameters

**To Reproduce**
Steps to reproduce the behavior:
1. Use two nRF52832 DKs
2. erase both boards using nrfjprog --eraseall
3. compile light switch proxy client & server examples from mesh sdk v2.1.1
4. download projects to the respective boards
5. open a debugging session in Segger Embedded Studio to see log output (no breakpoints)
6. Network tab -> Add new device
8. connect to client node
9. Do not make any changes to name, unicast address or appkey 1
10. press Provision

**Expected behavior**
I expect the nRF Mesh app to provision successfully.

**Important note:** When I exit the nRF Mesh app (press home button on iPhone), open the nRF Mesh app again & press "Reconnect", it seems the proxy client is provisioned correctly. This can also be seen from proxy client log above (<t:       1030>, main.c,  101, Successfully provisioned).

**Platform details:**
 - Device: iPhone 8
 - OS: iOS 11.4.1
 - Application version V1.0.1, Build Number 7

**Logs / Screenshots**
Add any logs or screenshots that can help us debug the issue faster

mostafaberg, 2018-07-31:
This has been resolved [here](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/commit/34d0a7a04cd20b1b9ac5147b71ef52110a6ce89f) and verified by other developers, for the time being, I'll add that to the next release and will verify/test once more.
Bug is related to the timings due to the connection params changes in the new Mesh SDK, What happens is all segments arrive, but the block acknowledgment's timer has not triggered yet, causing a timeout and the state breaks as you see. this happens only while doing an app key add message.

The new version of the iOS SDK and app are ready but not released yet, follow [The next release here](
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/projects/1)

jovirus, 2018-08-08:
I confirm the same issue, the provisioning process stopped at 91% after element models found. See the attached picture.
iPhone 7, 11.4.1
App model 1.0.1 build 7
[image]

mostafaberg, 2018-08-10:
Tested thoroughly in v1.0.2 and it does not reproduces after the fix has been in place, will close for now and open a new issue if the bug is still reproducing on other devices.

closed 2018-08-10 by mostafaberg