nordicsemi/IOS-nRF-Mesh-Library #83: cocoapods error installing OpenSSL dependency

macellari-mac, 2018-08-20:
Running 'pod install' I get an error during openssl dependency installation.

**To Reproduce**
run 'pod install'

**Expected behavior**
Verbose output terminating with:
cp: file.tgz: No such file or directory

It seems that the problem can be resolved changing the line:
    s.dependency 'OpenSSL'
to:
    s.dependency 'OpenSSL-Universal'
in the nRFMeshProvision.podspec file.

trusbe, 2018-08-21:
look here: https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/issues/23

mostafaberg, 2018-10-30:
This is fixed in the latest release, and the cocoapods is now released 👍!
to quickly test it out, please try to run the try command in your terminal
`pod try nRFMeshProvision`

closed 2018-10-30 by mostafaberg