nordicsemi/IOS-nRF-Mesh-Library #86: 91% issue is back again

macellari-mac, 2018-08-31:
**Describe the bug**
During node provisioning the process get stuck at 91% and stops.
The app needs to be terminated and the network db erased.

**To Reproduce**
Steps to reproduce the behavior:
1. Discovery new unprovisioned node
2. Change default name
3. Start provisioning
4. See error at 91%

**Expected behavior**
The provisioning process get stuck at 91%

**Platform details:**
 - Device: iPhone6s
 - OS: iOS11.4.1
 
**Logs / Screenshots**
[image]
[image]
[image]
[image]

philips77, 2018-08-31:
Hi, the author of this app is on vacations so it may take few weeks before you'll get a response. Sorry for that!

macellari-mac, 2018-08-31:
Hi Aleksander,

Mostafa is the only developer working on this iOS library?
Do you know when he will be back?

Lucio

A: NordicSemiconductor/IOS-nRF-Mesh-Library
Cc: Lucio Macellari; Author

philips77, 2018-09-02:
He owns the iOS part, yes. He'll be back in 2 weeks, as far as I know. Did you try nRF Mesh on Android? Do you have similar problem?

mostafaberg, 2018-09-28:
Issue is now resolved in the last PR, related to an TTL / acknowledgement issue that can only be resolved by a delay to wait for the retransmissions to be completed.

mostafaberg, 2018-09-28:
fix will be in the next release, for now it's on the develop branch 👍

closed 2018-09-28 by mostafaberg