nordicsemi/IOS-nRF-Mesh-Library #87: Notes

vikrant8051, 2018-09-07:
message should publish as set_unack (unreliable message).

   If it is unicast address, then it could be set message.

   In setting, give option for user to select whether he/she want to send reliable message or
   unreliable message in case of unicast destination address.

2) iOS APP observations:

2.1.
I've installed v1.0.2 build no. 9.
But not able to set various Publication parameters which are newly introduced feature.
If we set & retried to modify the things then App shows default values instead of previously entered values.

2.2
Mobile itself not properly re-transmit messages. If there are many NODEs in Group then NODEs which are in close vicinity of Proxy NODE are reacting properly, others are not since either Proxy is not re-transmitting or Mobile App is not re-transmitting.

3) Relay - On/Off (along with Relay re-transmission parameters settings)

4) Proxy - On/Off 

5) It will be great if Level GUI (seekbar which is going to introduced in next release under Generic Level Server ) publish real-time values instead of sending just last value.

Level Value range is -> -32768 to 32767 (16-bit 2's compliment)
Make sure that seekbar get divided into 10 virtual points which is hidden for user point of view
and if user scroll seekbar (from extreme left to extreme right) then only 10 values get publish.

If we don't create 10 discrete points then that could publish multiple values to Mesh network & that could hangs or degrade Mesh n/w performance.

Hope this helps to improve the things.
Thank You !!

vikrant8051, 2018-09-07:
Bluez v5.49 #meshctl supports following configuration commands .......

1. target <unicast>                                  Set target node to configure

philips77, 2018-09-07:
Hello! @mostafaberg is on vacations until 17th of September. He'll talk with you when he's back.
I'm the meantime you may test the Android app as its author is improving the app now. Thanks for all the feedback!

vikrant8051, 2018-09-07:
Hello @philips77 
Thanks for acknowledgment. I've already written email to @mostafaberg & @roshanrajaratnam.
In reply to it, Mostafa advised me to mentioned all these things via Github issues for his reference.

Thank you !!

mostafaberg, 2018-09-18:
Back at work 👍 will bee looking into this soon

philips77, 2018-09-28:
It's not possible to send values in real time when transition or delay times are greater then 0. I actually voted to have Send button at the bottom...

vikrant8051, 2018-09-28:
@philips77 @mostafaberg @roshanrajaratnam 
Send button at the bottom  :( 
Sorry but then current one is better.

I need real time value for testing purpose to check performance of Mesh n/w.
And believe me it will be helpful to improve or optimize firmware.

With real-time values firmware 
1. could crash
2. NODEs in single Group could behave asynchronously
3. Lag in response 
4. NODEs could take to mush of time to reach at final value if user vary slider very fast

For this to detect App should support to publish real time values (with max. 10 discrete virtual points on slider) & it should be handy & out-of-the-box in #nRFMesh like App.

philips77, 2019-08-20:
Hi, I'm closing this issue as it's regarding the old version of the app. As the second one is now under development I would ask you to test the new one if you could. It's not yet on App Store but should be in few weeks when it's feature-complete.

closed 2019-08-20 by philips77