nordicsemi/IOS-nRF-Mesh-Library #89: OpenSSL Dynamic Framework Support ？

trusbe, 2018-09-25:
I want to package this into a Framework based on this library but the OpenSSL library reports this error:

```
Ld: symbol(s) not found for architecture arm64
Clang: error: linker command failed with exit code 1 (use -v to see invocation)

```
Is it possible to provide an OpenSSL dynamic framework for easy integration into projects? or cocoapods ?

trusbe, 2018-09-25:
I used this to generate a static library of OpenSSL.

https://github.com/x2on/OpenSSL-for-iPhone

mostafaberg, 2018-09-28:
Still working on that, I think the only way to work around is to remove the dependency from the pod's dependencies, and add it to your application's podfile so it's built on your computer instead. until then there's not so much to do besides workarounds until we find a proper solution for OpenSSL :)

sorry for the inconvenience

closed 2018-09-28 by mostafaberg

phillippbertram, 2020-05-14:
is there already an update here? 🙃