nordicsemi/IOS-nRF-Mesh-Library #97: Provisioning process get stuck at 87%

macellari-mac, 2018-10-23:
**Describe the bug**
The provisioning process get stuck at 87% during the 12th node provisioning.
If I try to add a different node, other than the previous, the problem repeats.
The address (0x0017) is reused at each retry.

**To Reproduce**
Steps to reproduce the behavior:
1. Provision the first 11 nodes
2. Start provisioning os 12th node
3. The provisioning process get stuck at 87%

**Platform details:**
 - Device: iPhone6S
 - OS: iOS 12.0
 - Library Version: 1.0.4(12)

**Logs / Screenshots**
Screenshots:
[image]
[image]
[image]
[image]
[image]

mostafaberg, 2018-10-30:
Are you sure the confirmation value is correct ? a wrong pin code/ auth code will result in that

macellari-mac, 2018-10-30:
Hi Mostafa,
no pin code / auth code configured.
You think it could be a FW problem?
Same FW on all nodes but starting on 12th I cannot add more nodes with this error.

macellari-mac, 2018-11-07:
I verified that the issue does not occurs after a fixed number of provisioned nodes.
I reproduced the problem after 4 provisioned nodes. The fifth fails.
Something occurs on the app because when the issue occurs you are not able to provision more nodes, even different one from the first failed, you have to forget network.

mostafaberg, 2018-12-05:
Sorry for the late response I was off work, ok if you're sure it's not a auth issue then my second guess is clashing unicast addresses, what is the unicast for the provisioner ? and when you have that issue on a node, do you provision the next one with the same unicast ? (I see `0x0017` reused) this should happen, try manually setting the next one to `0x0017` + `0x0001` + the number of elements on that last node that crashed. and see if it works

most probably this is a bug in the apps state, but should work otherwise when you set your own unicast

macellari-mac, 2018-12-05:
The reuse of the unicast address is caused by the fail of the previous node provisioning/configuration.
Do you have any idea about what could be the cause of the first error at 87% ?

macellari-mac, 2018-12-07:
Today we provisioned 20 nodes for a test and the 87% issue occurred.
I confirm that if I manually change the address of the next node I can go on with the provisioning of other nodes.
The problem is that the node where the 87% issue occurred is corrupted and to be restored needs to be erased and flashed.

zanibonilighting, 2019-04-25:
Hi! is there any news about this issue? any hotfix we could use?

philips77, 2019-04-25:
Hi, I'm in a process of rewriting the library to adjust the model to the new JSON schema and make the lib transport-independent. It's a very early stage now. You may follow changes on `version2` branch. The current version will not be updated, but we are happy to accept PRs.

philips77, 2019-08-20:
Closing the issue as the old version is no longer maintained. Please create a new one if you find similar issue on the new version.

closed 2019-08-20 by philips77