nordicsemi/IOS-nRF-Mesh-Library #98: When I get the "meshState.bin" from A iphone and set the file to B iphone.

zhubin1987, 2018-10-24:
When I get the meshState.bin from A iphone and set the file to B iphone.
The data is ok, appkeys and network key is same.but B iphone can not controll well.A iphone is OK
So what may be the problem? Thanks for give me some suggest

trusbe, 2018-10-24:
Mobile phone unicast address must be different.

If the unicast addresses of the two phones are the same, you need to ensure that the nRFMeshSequenceNumber is incremented.

@zhubin1987

mostafaberg, 2018-10-30:
@zhubin1987 exactly, as @trusbe recommends, if both phones have the same unicast things will not work, since the sequence numbers will be invalid and probably lower than previous values which will lead to messages being always dropped.

the fix is to simply copy over the state file, and then go to settings and change the unicast address of the phone to a new unused value, after that things should work as expected :)

Thanks for the help @trusbe !

closed 2018-10-30 by mostafaberg

zhubin1987, 2018-11-02:
@trusbe @mostafaberg  thanks for your answers. It works well with both of your two ways.Thank you