nordicsemi/IOS-nRF-Mesh-Library #108: Generic Level model values range goes only until about 3%

zanibonilighting, 2019-01-11:
Hi,
When using the iOS app to control the dimming example on meshSDK/examples/experimental_dimming/dimming server the range goes from 0% to about 3% only then setting level to 100% (we changed the pin polarity btw, but the same happens on the stock example).
This happens only on the iOS app, the android app works ok with the generic level.
iOS app:
[image]
Android app:
[image]
Thanks!

zanibonilighting, 2019-01-15:
Hi again!

So I found a weird conversion on ⁨IOS_App⁩ ▸ ⁨Example⁩ ▸ ⁨nRFMeshProvision⁩ ▸ ⁨ViewControllers⁩ ▸ ⁨NodeConfigurationView⁩ ▸ ⁨ModelConfigurationView⁩ ▸ ⁨didChangeSliderOnCell line 948:

var level = Int(((newSliderValue * 65535) / 100) - 32768)

I don't see why the / 100… anyways I removed and now it seems we have a full range from -32768 to +32768… but I dont know why the developer made that conversion and I dont know which test to run so we cant be sure I didn’t just bring a tons of bugs… so I think to leave this question open for now until someone with a bit more experience with the app can confirm the hot fix...

roshanrajaratnam, 2019-01-16:
Hi @zanibonilighting thanks for reporting this issue I see what the problem here is. The slider value is from ranges from 0.0 - 1.0 so I guess the mistake was the division by 100 forgetting the slider does not range from 0-100. Nice catch!

zanibonilighting, 2019-01-16:
awesome!

roshanrajaratnam, 2019-01-16:
I would leave it open until we fix this, although we welcome pull requests. If you would like you can contribute towards the library! All you have to do is to sign the CLA when sending a pull request :tada:

philips77, 2019-08-20:
Closing the issue as the old version is no longer maintained. This should work fine in the new version.

closed 2019-08-20 by philips77