nordicsemi/IOS-nRF-Mesh-Library #110: receivedModelPublicationStatus and receivedModelSubsrciptionStatus called multiple times

bhavik-volansys, 2019-02-01:
**Describe the bug**
We are trying to create a group and for that, we need to subscribe and publish to some particular group id like "C001" for on/off model. When we publish to id "C001" the delegate method is calling almost 3-4 times and it's not fixed so is there any way that we can find some index or something to find the last packet or any other workaround?

And

We want to subscribe and publish on both models on/off and level so we have to make total of 4 calls for a single device. Publish/subscribe for on/off [0x10, 0x00] and publish/subscribe for level [0x10, 0x02]. If we want to add 10 devices in C001 then we have to call 40 calls. 

Please guide us for these..

philips77, 2019-02-04:
Hi,
Few questions:
1. Is the delegate called once after subscribing only one device, twice after the second etc.?
2. Do you get those callback during subscription or when sending a message to the group? Are you publishing acknowledged messages? Perhaps you are getting a callback from each node that had received the command?
3. Are you using the sample app or trying to include the lib in your project? 
4. Did you do any further debugging and have more information?

Sorry, I can't help more at the moment :( I'm not the author of this library.

bhavik-volansys, 2019-02-04:
1. Is the delegate called once after subscribing only one device, twice after the second etc.?
> Not, it is calling delegate method multiple times for single device only

2. Do you get those callback during subscription or when sending a message to the group? Are you publishing acknowledged messages? Perhaps you are getting a callback from each node that had received the command?
> During subscription.

3. Are you using the sample app or trying to include the lib in your project?
> I have tried on both the application. Sample app and also integrated into our app

4. Did you do any further debugging and have more information?
> If you want then I can share the logs but that will be encrypted.

philips77, 2019-02-04:
Thanks,
I talked with the author of this library, who unfortunately has transition to new tasks, and he suggests that the device may send multiple responses to subscription request (retransmission?). Perhaps the phone does not ack that it received the message?
Until we assign more resources to the app I sorry but I can't help you more. Could you dig on your own and prepare a PR with possible fix?

bhavik-volansys, 2019-02-11:
Hi,
We are kind of stuck now, we have tried to find a way to check for the last packet but we are not able to find that. We just want to know that how we can identify that delegate method `receivedModelPublicationStatus` will not call now?

philips77, 2019-08-20:
Closing the issue as the old version is no longer maintained. Please try the new version which, I hope, does not have this issue.

closed 2019-08-20 by philips77