nordicsemi/IOS-nRF-Mesh-Library #112: Bug with multiple generic level server models on same node

ManishSudini, 2019-02-07:
**Description**: I added two different generic level servers on two different elements to the experimental_dimming -> dimming_server example to control two different LED's on the development board. I provisioned the board using the iOS app. When I control the server model 1 on the app, it is controlling the LED 1.But, when I control the server model 2 on the app, it is also controlling the LED 1 whereas it should be controlling LED 2.
I tried the same example with the Android app and I was able to control both LED's successfully.
This might be an issue in the iOS app that when various generic level server models try to publish the information onto the board, there is an issue with the message that is being received by the board from the iOS app as messages from various models(similar) are getting routed to the address of the first model.

**To Reproduce**
Steps to reproduce the behavior:
1.  Create another generic_level_server model in the dimming_server example with specific functions to control a different LED on the Development board.
2. Provision the board using the iOS app
3. Try setting the level on the generic level server 1 and 2.
4. Error: Generic level server model 2 should control LED 2 whereas it controls LED 1.

**Expected behavior**
Generic level server model 1 should control LED 1.
Generic level server model 2 should control LED 2.

**Platform details:**
 - Device: IPad, IPhone 5
 - OS: iOS12.1.3
 - Library Version 1.0.4

philips77, 2019-02-08:
Thank you for reporting the issue. We will look into it, but nRF Mesh for iOS has been put on hold for a while, unfortunately. We will come back to you. 
For the time being please use the Android nRF Mesh as it's in continuous development.

PaulLeGuennec, 2019-03-01:
Hello, 
Do you have an idea of when the development of the iOS app will resume ?
My app is stuck during provisioning.
Thanks for your response.

philips77, 2019-03-01:
Hello,
I can't promise you any release date right now. I will be responsible for it from the next sprint, but I have to learn a lot before I start adding features.

PaulLeGuennec, 2019-03-01:
Ok, thanks for the response. I will use an Android device for now.

philips77, 2019-08-20:
Hi, we are almost ready with the second version of the library and the sample app. It's not yet on App Store, however. I think what you described should work in the new version, please give it a spin if you have a Mac and Xcode to test. I'm closing this issue, if you find it not working with the new version, please create a new issue. Thanks!

closed 2019-08-20 by philips77