nordicsemi/IOS-nRF-Mesh-Library #116: Individual controls controlling multiple elements

SamS3lly, 2019-03-04:
**Describe the bug**
Not sure if this is a bug or it should work like this, summary: having 2 or more nodes, if we configure both/all elements and models to the same group address and then control one of the elements with the on/off button or the generic level slider, all nodes subscribed to the same group with change their state as well, is this suppose to happen?

**To Reproduce**
Steps to reproduce the behavior:
provision 2 or more devices
configure the same group address on the same models on all the elements (ie on/off model or generic level model)
turn on / off one of the elements or change the slider
all the elements configured to subscribe to the same group address will turn on / off

**Expected behavior**
shouldn't only the individual element be affected with the on / off? the only time that all the elements should respond to the on / off command is when sending that command to the group address?

**Platform details:**
 - Device: iPad
 - OS: iOS 12
 - 1.0.4

**Logs / Screenshots**
Add any logs or screenshots that can help us debug the issue faster

philips77, 2019-03-05:
It's like this until we implement proper group Control. There was an update of Android nRF Mesh today, please use Android until this is also implemented on iOS.

philips77, 2019-08-20:
Closing the issue as the old version is no longer maintained. In the new version the message is sent to individual element by its Unicast Address. The Group control is not yet complete, but should be in coming days.

closed 2019-08-20 by philips77