nordicsemi/IOS-nRF-Mesh-Library #117: Groups implementatiomn

SamS3lly, 2019-03-04:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

philips77, 2019-08-20:
This feature is coming soon in the currently developed second version of the lib/sample app. I'll close the issue when it's finished.

philips77, 2019-09-06:
Group controls are now working. From the sample app we implemented only Generic OnOff and Generic Level models, but they are enough for now.

Answering your questions:

> should this "item" be a whole element with all its models? or should every model be added separately?

You subscribe models to groups, so we decided to display them. If there are multiple App Keys bound to such model, we also allow users to send messages using selected one of them, as they may target different devices. For example all Generic OnOff may be in the "Kitchen" group, you you control lights and power plugs separately.

> lets say that we have a server with an on/off model and generic level model, should both models be added to the group automatically or should every model be added separately?

You add each model separately, at least no the very low level, where you send messages. How you display it in your app for the user it's a different story and you should focus on best UX.

closed 2019-09-06 by philips77