nordicsemi/IOS-nRF-Mesh-Library #118: Wrong PDU, expected 0306

trusbe, 2019-03-26:
When the device is connected to the network, an error is reported: Wrong PDU, expected 0306

Excuse me, what is this question?

Xcode: 10.2 (10E125)
Swift: 4.2

roshanrajaratnam, 2019-03-26:
@trusbe It seems to be happenning during the provisioning process. Once the provisioning confiirmation is sent the app is supposed to receive the provisioning random confirmation. However there seems to be something else that's received hence causing the disconnection.

trusbe, 2019-03-27:
@roshanrajaratnam  I used Xcode to go back to 10.1, and it will run normally. Let's develop it with 10.1 for the time being. Wait until there is time to see what the problem is.

zickezacke, 2019-06-10:
Hi,

I'm experiencing the same problem and can't work with 10.1 anymore. Did somebody fixed the problem or knows another work around?

trusbe, 2019-06-18:
@zickezacke  You can refer to this solution  
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/issues/120#issuecomment-499032642

zickezacke, 2019-06-18:
> @zickezacke <https://github.com/zickezacke> You can refer to this solution
> #120 (comment)

Thanks for your reply.

I had already tried this solution and it worked.

Best regards,

Ronny

philips77, 2019-08-20:
Closing the issue as the old version is no longer maintained. Please migrate to version 2.

closed 2019-08-20 by philips77