nordicsemi/IOS-nRF-Mesh-Library #119: I rewrote the code with OC and found some strange phenomena.

DistanceLe, 2019-03-29:
Chip model: nRF52832

When I use this demo（swift）, I find that after the device provisioning is complete, the peripheral will switch the service (1827->1828), and the process will last for about 10 seconds.
But when I rewritten with OC, I found that switching services took less than 1 second.

I have a question now, the peripheral switching service should be the operation of the chip, it should be independent of the App, but now this phenomenon is somewhat unclear.
(I also used the Android demo, this switching process lasted about 6 seconds)

philips77, 2019-03-29:
I guess it's just a matter of when the phone finds out that it got disconnected. Without graceful disconnect it may take a while, depending on the connection parameters, system and perhaps something else. The device needs the same time to store all settings and reboot in every case.

closed 2019-08-20 by philips77

songhongkang, 2021-09-09:
oc 代码能开源吗

songhongkang, 2021-09-09:
@DistanceLe

DistanceLe, 2021-09-09:
@songhongkang 
我这个是给公司用的，不好开源。 
当时他这个demo bug比较多才重写的。
Nordic 后面自己也重写了一个版本，现在最新的版本功能很全了。