nordicsemi/IOS-nRF-Mesh-Library #120: Xcode 5 update causes incorrect random pdu

SamS3lly, 2019-04-01:
After installing Xcode 5 I get an incorrect 'random pdu' error: "Wrong PDU, expected 0306".
The random PDU is created successfully and is correct but for some reason once passed to 'RandomConfirmationProvisioningState.swift's CBManager didUpdate func it returns 0309 instead

**To Reproduce**
Download the new version of Xcode and provision a node.
half way through provisioning will fail and say "Wrong PDU, expected 0306".

**Platform details:**
 - Device: Ipad Air A1474 
 - OS: 12.1.4

trusbe, 2019-04-02:
Same question.

https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/issues/118

philips77, 2019-04-02:
hi, I'm working on a newer version of the library, with modified model (and API). You can see a glimpse on branch `version2`, but it's an initial work. I will restore some files from the master branch if I find them necessary, but most of the app will be written from scratch. I'll make sure the issue is fixed in the new version. It will take some time before the initial version will be released.

zanibonilighting, 2019-04-24:
Hi!
Is there a way to have a hotfix? or at least some clues to where to look for this? its kind of hard to find what would have being the bug introduced when migrating to XCode5....

trusbe, 2019-04-25:
@zanibonilighting  that's true. hope to give priority to this problem.

bozhi, 2019-05-04:
Problem is with closures.

philips77, 2019-05-04:
By xcode 5 you mean xcode 10.2 and Swift 5?

trusbe, 2019-05-05:
@philips77  right.

jnicklas, 2019-05-24:
I'm also having this problem right now, and it's a bit of a showstopper for me, I can't get past provisioning to continue developing my app. I've tried to downgrade XCode and iOS, but haven't been able to make it work. A hotfix for this would be really great, or if you have some insight into what is going wrong then maybe I can fix it myself.

zanibonilighting, 2019-05-24:
Hi Jonas, can I ask what’s the issue when downgrading? For us it was: some of the guys deleted Xcode 10.2, ( I kept it… not a problem ), then download the 10.1 from here: https://developer.apple.com/download/more/ <https://developer.apple.com/download/more/> then I guess your devices has the latest iOS version? In that case you would need the missing support files from here: https://stackoverflow.com/questions/55575782/xcode-missing-support-files-ios-12-2-16e227/55576391 <https://stackoverflow.com/questions/55575782/xcode-missing-support-files-ios-12-2-16e227/55576391> and you should be good to go I think..? If you get some other errors, more info would be useful.

In any case, since the V2 of the iOS app is in progress… the more Aleksander needs to support V1 the longer it takes to get the V2… and I think we all can agree that V2 is looking waaaaayyy nicer than V1… and V1 will be obsolete as soon as V2 covers the basics… so for now we are coping with V1 + Xcode 10.1 and migrating to Xcode 10.2 is not a priority for us at all anymore… since we will be adopting V2 as soon as its foundations / basics are usable.

jnicklas, 2019-05-27:
@zanibonilighting thank you, this was very helpful! I'm not really an iOS developer so not too familiar with the tooling and this put me on the right track to making XCode 10.1 work with iOS 12.2. With some additional fiddling I was able to get through the whole provisioning process 🎉 

I did look at the new version and it looks quite a lot better than what we currently have, which is exciting. I do agree that focusing on the new version is better, especially if it is not too far off.

macellari-mac, 2019-06-04:
@zanibonilighting I would ask you to take into account that many of us, including myself, have apps almost in production based on version 1 of the library/app.
We are not only working on nordic ble mesh on our company and we are stuck to Xcode 10.1 because of this issue.
With all respect I feel that it cannot be Nordic to decide whether to migrate to new xcode versions is a priority or not for us. Please understand what I mean.
Besides that, I am not sure that when you will release the V2 libraries we will have available resources to immediately implement the update from V1 to V2.
A hotfix at least for this issue would be welcome.

zanibonilighting, 2019-06-04:
Hey @macellari-mac please note that we are not Nordic, this issue was opened by us (@SamS3lly is part of us), and I said that its not a priority for us... and we don't set priorities for Nordic... and just keep in mind, we didn't closed the issue either, bottom line: for US this is not a priority at all but its still an issue so we kept it open, and in our opinion, its way more important to get V2, the V1 was always a beta version of the mesh library, it was never released as a fully implemented library as much as I know.

My advice would be the same we are doing at the moment, prepare everything that we have done on V1 for migration to V2, any new implementation / functionality will happen on V2, so IMHO staying on V1 is not the best strategy at all...

The real true question for us is, @philips77 any ideas about a general roadmap? when we can expect a usable V2 provisioning app? at least at the same level of V1?

philips77, 2019-06-05:
Have a look at #126.

mixaillanatolich, 2019-06-05:
The issue in method _func leftPad(length: Int) -> Data_ in _Data+utils.swift_

I've changed 
        let bytes = self.withUnsafeBytes { (aByte) -> UnsafeRawPointer in
            return UnsafeRawPointer(aByte)
        }

to

        let nsData = self as NSData
        let bytes = nsData.bytes

And the issue  "Wrong PDU, expected 0306" was fixed for me

trusbe, 2019-06-12:
@mixaillanatolich  I am running normally here. 

It is recommended that all tests that have this problem be tested. If it is fixed, it is recommended to submit a PR.

PM-BubblyNet, 2019-06-19:
Applied the fix and tested, working well so far, I mean the 87% crash on about 30% of the provisioning attempts is still there but this wrong PDU issue seems resolved, @BubblyNetDev please close this one :)

closed 2019-06-24 by BubblyNetDev