nordicsemi/IOS-nRF-Mesh-Library #121: Send acknowledged and unacknowledged messages

SamS3lly, 2019-04-08:
Right now the generic set messages are writing characteristic values withoutResponse, does this mean the nodes are sending Unacknowledged messages?

closed 2019-04-08 by SamS3lly

d-srd, 2019-06-19:
@BubblyNetDev Could you please provide an update regarding this issue? Is the library sending acknowledged or unacknowledged messages? Are you somehow able to control that?

philips77, 2019-06-19:
Hi,
The main branch of this library is no longer maintained. Please, have a look into branch `version2`, which is in heavy development. 
All messages are sent using `.withoutResponse`, but that is a requirement for the Proxy Service. Whether a message is acknowledged or unacknowledged depends on 2 things: 
1. segmented messages are acknowledged (although the lib does not report the success/fail to the app yet)
2. there are also ack on Access Layer, which are delivered to the app, for example a response for `ConfigCompositionDataGet` is `ConfigCompositionDataStatus`.