nordicsemi/IOS-nRF-Mesh-Library #125: Mesh State is unable to recover if configure step fails

netzon-service, 2019-05-03:
**Describe the bug**
When node.configure() function fails, MeshState.provisionedNodes already has a new entry but nextUnicast was not updated. This will result in using the wrong device Key for each succeeding provision.

**To Reproduce**
Steps to reproduce the behavior:
1. Start to Provision a device
2. Connect to the device. // node.connect()
3. Identify the device. // node.identifyWithDuration()
4. Provision the device. // node.provision()
4.5 Turn off the device 
5. Configure the device. // node.configure()
6. Configure fails.
7. Succeeding provision will now fail due to device key.

**Proposed Fix**
state.incrementUnicastBy() in CompositionGetConfiguratorState.receivedData() should be transfered to MeshProvisioningDataTableViewController.nodeProvisioningCompleted() when state.provisionedNodes.append(nodeEntry!) happens.

The number of elements should be extracted from InviteProvisioningState.peripheral().InviteCapabilities.

**Platform details:**
 - Device: iPhone6
 - OS: 12.1
 - Master branch (58e5ed76a8fda9972e31cfeec81543ae27ff5646)

**Logs / Screenshots**
unprovisioned node connected
starting service discovery
started discovery
Discovered mesh provisioning service
Discovering characteristics for provisioning service
Discovered data in charcateristic
Discovered data out characteristic
discovery completed
deviceKey() calculatedDeviceKey: Enil
nodeDidCompleteDiscovery Dimmer nil
Connecting Completed
Start Identifying
030001
provision invite data: 0x01
Received capabilities provisioning message
Element count: 4,Algorithm: fipsp256EllipticCurve, PublicKeyAvailable: publicKeyInformationUnavailable, StaticOOBAvailable: staticOutOfBoundInformationAvailable, OutputOOBSize: 0, InputOOBSize: 0
No output actions supported
No input actions supported
capabilities : 0x0400010001000000000000
deviceKey() calculatedDeviceKey: nil
nodeCompletedProvisioningInvitation(_:withCapabilities:) Dimmer nil
Start Provisinoning
2019-05-01 12:24:35.997: rx_provisionNode -> Event next(true)
Executing Start provision PDU
Provision Start PDU Sent: 03020000000000
provision start data: 0x0000000000
switched provisioning state: Public Key
Executing public key provision PDU
generating keypair
keypair generated
keypare generated, pubkey: 0xF1413020DA4920397067606417265181A65CA40BEBBD849DEBF23F3EAEFA29128F8B4D8158D855988D881362D6AF61BC91DF03CCA71A4AC2AA1590B2FF94821C
No SAR needed, maximum write length for write without response 66
provision done
Received an unsegmented device public key
Public Key = C584C969D004E129BD66F0990FF64B3A590C083722A1C02FC49D63446E7418A19DF67270412614F0FB60F29E9DDF59B082A5DF6AE57970F8FB2F0AC9257E565F
device public key: 0xC584C969D004E129BD66F0990FF64B3A590C083722A1C02FC49D63446E7418A19DF67270412614F0FB60F29E9DDF59B082A5DF6AE57970F8FB2F0AC9257E565F
calculated DHKey: 0xC67464DBB8F301F94F137B5BBA90EC3AF45588EB56B9F507B1EC5A94F117BEF3
switched provisioning state: Provisioning confirmation
No OutputOOB capabilities
confirmationInputs: 0104000100010000000000000000000000F1413020DA4920397067606417265181A65CA40BEBBD849DEBF23F3EAEFA29128F8B4D8158D855988D881362D6AF61BC91DF03CCA71A4AC2AA1590B2FF94821CC584C969D004E129BD66F0990FF64B3A590C083722A1C02FC49D63446E7418A19DF67270412614F0FB60F29E9DDF59B082A5DF6AE57970F8FB2F0AC9257E565F
Salt: FD23038DFE358C4AB790F11C166CFCDA
ecdh: C67464DBB8F301F94F137B5BBA90EC3AF45588EB56B9F507B1EC5A94F117BEF3
confirmationKey: B91E0C9A5A1BB4B14B40391C780B1278
input complete: 0x00000000000000000000000000000000
provisioner confirmation: 0x91B5D502666D9FB9E120DF05ADD1069F
ConfirmationVal: 91B5D502666D9FB9E120DF05ADD1069F
provisioner random: 0x74DFD4E471E583D7DF1CA5F75FA1113A
randomProv: 74DFD4E471E583D7DF1CA5F75FA1113A
Confirmation PDU sent: 030591B5D502666D9FB9E120DF05ADD1069F
provisioner confirmation: 0xBD53D970CBD3EF22133D09B894D1476F
switched provisioning state: Random confirmation
device random: 0xAFDEAA5D0E2E2D0FA55A6D5528C56E6A
switched provisioning state: Provisioning data
calculatedDeviceKey aDeviceKey: 6CC156E758A045D4AB1DC7DD3EF3A1CF
Target Name          = Dimmer
Target UUID          = 6532A30F-F4A0-26E8-6575-0EC3EDB31AF9
Provisioning Salt    = 8DA4446CB38EC01BD4988D856303D554
Session Key          = 3FE0CEB30BCEEB3405D54AA5C5EE9D53
Session Nonce        = 48818151983689ED4765465DAA
NetKey               = 2D13414F9F2D5874AE396C2A71E39699
DeviceKey            = 6CC156E758A045D4AB1DC7DD3EF3A1CF
KeyIndex             = 0000
Flags                = 00
IV Index             = 00000000
Unicast Address      = 0035
Provisioning data: 2D13414F9F2D5874AE396C2A71E39699000000000000000035
Provisioning data PDU: 0307C9570AE5C343F6D48761E49F083ABD5584B588EB6FBF17393BD8FF6FC6100E304B
No segmentation necessary to send provisioning data, write length = 66
0308
Provisioning succeeded, will delay 1 second to check if service change
No services chagned, a reconnect is needed
deviceKey() calculatedDeviceKey: Optional("6CC156E758A045D4AB1DC7DD3EF3A1CF")
nodeProvisioningCompleted Dimmer Optional("6CC156E758A045D4AB1DC7DD3EF3A1CF")
Provisioning succeeded
deviceKey() calculatedDeviceKey: Optional("6CC156E758A045D4AB1DC7DD3EF3A1CF")
getNodeEntryData() deviceKey: 6CC156E758A045D4AB1DC7DD3EF3A1CF
deviceKey() calculatedDeviceKey: Optional("6CC156E758A045D4AB1DC7DD3EF3A1CF")
nodeShouldDisconnect Dimmer Optional("6CC156E758A045D4AB1DC7DD3EF3A1CF")
2019-05-01 12:24:38.054: rx_nodeProvisioningCompleted -> Event next(true)

disconnected 1
found proxy node with node id: 0182225E7786F3568B8811BE9F349539FC
verifying NodeID: 0182225E7786F3568B8811BE9F349539FC
node identity verified!
unicast found: 0035
Connecting Devices
provisioned proxy node connected
starting service discovery
Discovered mesh proxy service
Discovering characteristics for provisioning service
Discovered data in charcateristic
Discovered data out characteristic
nodeDidCompleteDiscovery
isAppKey: false
opcode:   8008
payload:  FF
key:      BE22EF562051B22F35AB0F84B03B6FED
netKey:   2D13414F9F2D5874AE396C2A71E39699
src:      0123
dst:      0035
ivIndex:  00000000
seq:      SequenceNumber(count: 491, defaultsKey: "nRFMeshSequenceNumber")
ttl:      04
encrypted upperTransport PDU: 7BFC1648438335
decrypted upperTransport PDU: nil
raw upperTransport PDU: Optional("8008FF")
Ready to send 1 payloads
Composition get message to set:0061152E140147CDB61FD4BC367A8910594869D45FBB
Cahrcateristic value updated: 0101006362A0EA489CD3EB0000000011D272A35DDAC42B
Secure beacon: 0101006362A0EA489CD3EB0000000011D272A35DDAC42B

netzon-service, 2019-05-17:
I guess this is the same as https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/issues/53#issue-345150242

philips77, 2019-05-17:
Hi, I'm working on 2nd version of the library and sample app. They are on version2 branch. So far only provisioning is working, but it's in constant development.

robertsmedshammar, 2019-05-24:
@philips77 , it sounds great that you are working on a new version of the mesh iOS mesh library and that it's now in constant development. When do you think the library is ready to be used where we can add models on top of it? I know that the usual answer is to contact sales representative, but it would be interesting just to hear what will be added in the next coming 2 sprints or 2 months.

philips77, 2019-05-24:
As you may have seen, provisioning is working now, with all oob types. The next step is doing configuration, which requires finishing almost all the layers. I hope adding more things to the lib will go smoothly after that. It's just a matter of polishing the ui in the sample app, i think.

philips77, 2019-06-20:
Please, follow on #126.

closed 2019-06-20 by philips77