nordicsemi/IOS-nRF-Mesh-Library #126: [QUESTION] V2 general availability

d-srd, 2019-06-04:
Hi!

I was just wondering - when do you think that V2 will be ready for general public usage?

philips77, 2019-06-05:
Depends on what you want in it. I'm working on implementing layers now. Next thing is configuration and this will make the app "usable". But there is much more that is already supported on Android library: generic and vendor messages, group management, proxy filter, just to mention mesh features. 
I work 100% of time now on this library and sample app. I expect it to be functional this month, but I'm an optimist :)

zanibonilighting, 2019-06-10:
hey @philips77 thanks for the info, I went on, pulled v2 and build it, identifying happens fine but then when clicking on "provision" the app crashes... :(

Connecting...
Connected
Discovering services...
Service found
Discovering characteristrics...
Data In characteristic found
Data Out characteristic found
Enabling notifications...
Data Out notifications enabled
GATT Bearer open and ready
New state: Provisioning Invitation sent
-> 0x030005
<- 0x03010400010001000000000000
New state: Provisioning Capabilities received:
--- Capabilities ---
Number of elements: 4
Algorithms: FIPS P-256 Elliptic Curve
Public Key Type: None
Static OOB Type: Static OOB Information Available
Output OOB Size: 0
Output OOB Actions: None
Input OOB Size: 0
Input OOB Actions: None
--------------------
2019-06-10 10:58:58.999116-0400 nRFMeshProvision_Example[1156:500932] *** Terminating app due to uncaught exception 'NSGenericException', reason: 'Your application has presented a UIAlertController (<UIAlertController: 0x10487c400>) of style UIAlertControllerStyleActionSheet from UINavigationController (<UINavigationController: 0x105067400>). The modalPresentationStyle of a UIAlertController with this style is UIModalPresentationPopover. You must provide location information for this popover through the alert controller's popoverPresentationController. You must provide either a sourceView and sourceRect or a barButtonItem.  If this information is not known when you present the alert controller, you may provide it in the UIPopoverPresentationControllerDelegate method -prepareForPopoverPresentation.'
*** First throw call stack:
(0x1c354a758 0x1c2752d00 0x1efa7bf70 0x1efa84b54 0x1efa825a8 0x1efb661e8 0x1f0132aa8 0x1f0120c70 0x1f014f078 0x1c34d9ee8 0x1c34d49e0 0x1c34d4f40 0x1c34d4764 0x1c5710dd4 0x1f012700c 0x102719ef0 0x1c2f93ffc)
libc++abi.dylib: terminating with uncaught exception of type NSException
(lldb)

philips77, 2019-06-20:
**Status update:**
Composition Data and Reset are working now.
It's still a bit unclear when they can work, as there is no indication for connection state yet, but by looking at the console it's clear :)

fajdof, 2019-07-02:
Hi, so basically what we need is provisioning of the network, reseting nodes (removing them from the network) and generic vendor messages. Do you have a rough estimate of when a version containing these will be released?
Thanks,
Filip

philips77, 2019-07-02:
I'm working now on binding app keys. Next step publish, subscribe and vendor messages. 
I'm going on vacations starting next Monday, but hopefully those will work, at least in the lib, if i can't make it in the sample app.

philips77, 2019-07-26:
Publications and subscriptions should work now. Groups are implemented, but there is no UI for controlling them in the sample app yet.

fajdof, 2019-07-26:
@philips77 thanks for the update!

philips77, 2019-08-20:
Vendor messaged should be operational now (in the library and individual controls in Node->Element->Vendor Model). Still some unit tests would be useful, but the manual tests using UI are passing.

philips77, 2019-08-20:
Coming next are: 
* Generic On/Off messages in library and UI
* Perhaps some more models and messages
* Group control.
The app will go live on App Store when those are ready. Next set of features will go to 2.1.

philips77, 2019-08-23:
Generic On/Off controls are ready.

PM-BubblyNet, 2019-08-23:
Wow Awesome! Thanks! and just to know... are you still on holidays? :)

philips77, 2019-08-23:
Hehe, no, I'm not.
Btw, today new set of generic messages have been added, only to the library.
Here's the list of supported generic models:
* Generic OnOff
* Generic Level
* Generic Default Transition Time
* Generic Power OnOff
* Generic Power OnOff Setup
* Generic Power Level
* Generic Power Level Setup
* Generic Battery

philips77, 2019-09-06:
New features added in the last 2 weeks:
1. Group controls (for Generic OnOff and Generic Level),
2. Individual controls for Generic OnOff, Generic Level and Vendor Model models.
3. Proxy Filter - upon connection to a Proxy, it will automatically subscribe to all unicast addresses of all elements set on the local device, all group addresses that any local model is subscribed to and to All Nodes groups address (0xFFFF). Filter may be configured using `ProxyFilter` class.

philips77, 2019-09-06:
FYI: Next feature that I plan to add is Proxy Filter tab, that will allow to control proxy filter and will display incoming messages. I think I'll also improve logging as a part of this task.

BubblyNetDev, 2019-09-09:
When do you think you will be implementing the enabling and disabling of Proxy in the Example app?

philips77, 2019-09-09:
It should be ready this week.

philips77, 2019-09-13:
Hi, sorry, but I didn't manage to finish (or even start) UI for controlling Proxy Filter. I'll do this next week. But the logging part is now ready! Check the [Console](https://support.apple.com/en-gb/guide/console/welcome/mac) and filter by subsystem "com.nordicsemi.nRFProvisioner".

philips77, 2019-09-13:
[image]

philips77, 2019-09-20:
Resending and timeouts for Acknowledged Mesh Messages is complete. You may tailor the intervals in the manager. They should rely on the number of segments sent and received, the distance to the target node, the TTL, connection interval, etc. It's hard to estimate the right retry interval. Right now I set it to the base value + 50 ms * TTL + 50 ms * number of segments of the request.
The problem is that often the request is short, but the response is long, or the other way round. But this cannot be estimated correctly.

For group messages it is enough to receive one response to cancel the timeout.

philips77, 2019-10-29:
All features I wanted to implement are complete. I'm about to publish the version on CocoaPods today. To be honest, there is an important feature that hasn't been implemented, which is the IV Update. IV Updates are handled when the update is coming from a remote node, but the provisioner will not initiate IV Index even if its sequence numbers are getting close to rollover.
Also, the Key Refresh procedure is not available from the sample app, but should be doable using the library.

philips77, 2019-10-30:
2.0.0 has been released. The sample app should shortly be available on AppStore.
All future feature requests will be addressed in future versions.
I'm closing this issue now.

closed 2019-10-30 by philips77