nordicsemi/IOS-nRF-Mesh-Library #127: V2 provisioning crashes

zanibonilighting, 2019-06-10:
hey @philips77, I went on, pulled v2 and build it, identifying happens fine but then when clicking on "provision" the app crashes... :(

debugger output:

Connecting...
Connected
Discovering services...
Service found
Discovering characteristrics...
Data In characteristic found
Data Out characteristic found
Enabling notifications...
Data Out notifications enabled
GATT Bearer open and ready
New state: Provisioning Invitation sent
-> 0x030005
<- 0x03010400010001000000000000
New state: Provisioning Capabilities received:
--- Capabilities ---
Number of elements: 4
Algorithms: FIPS P-256 Elliptic Curve
Public Key Type: None
Static OOB Type: Static OOB Information Available
Output OOB Size: 0
Output OOB Actions: None
Input OOB Size: 0
Input OOB Actions: None
2019-06-10 10:58:58.999116-0400 nRFMeshProvision_Example[1156:500932] *** Terminating app due to uncaught exception 'NSGenericException', reason: 'Your application has presented a UIAlertController (<UIAlertController: 0x10487c400>) of style UIAlertControllerStyleActionSheet from UINavigationController (<UINavigationController: 0x105067400>). The modalPresentationStyle of a UIAlertController with this style is UIModalPresentationPopover. You must provide location information for this popover through the alert controller's popoverPresentationController. You must provide either a sourceView and sourceRect or a barButtonItem. If this information is not known when you present the alert controller, you may provide it in the UIPopoverPresentationControllerDelegate method -prepareForPopoverPresentation.'
*** First throw call stack:
(0x1c354a758 0x1c2752d00 0x1efa7bf70 0x1efa84b54 0x1efa825a8 0x1efb661e8 0x1f0132aa8 0x1f0120c70 0x1f014f078 0x1c34d9ee8 0x1c34d49e0 0x1c34d4f40 0x1c34d4764 0x1c5710dd4 0x1f012700c 0x102719ef0 0x1c2f93ffc)
libc++abi.dylib: terminating with uncaught exception of type NSException
(lldb)

PM-BubblyNet, 2019-06-19:
hey! @philips77 one of our developers found the issue, we use iPads and iPhones and this happens only on iPads, and he got this hot-fix on SettingsViewController.Swift, line 140:

        // iPad actionSheet fix.
        if UIDevice.current.userInterfaceIdiom == .pad {
            alert.popoverPresentationController?.sourceView = self.view
            alert.popoverPresentationController?.sourceRect = CGRect(x: self.view.bounds.midX, y: self.view.bounds.maxY, width: 0, height: 0)
            alert.popoverPresentationController?.permittedArrowDirections = [.down]
        }

now provisioning button doesn't crash the app...

philips77, 2019-06-20:
The issue was in more than one place. I hope I fixed all instances. 
Also, I dig my iPad from a pile of dirt, so I'll now be testing against iPad as well :) At least from time to time.

PM-BubblyNet, 2019-06-24:
hey, found another one, when clicking "Export" (Config) "Import" don't have the issue.

BubblyNetDev, 2019-06-24:
Same fix as before this time in the exportNetwork function in the settingsViewController.

`                    if UIDevice.current.userInterfaceIdiom == .pad {
                        controller.popoverPresentationController?.sourceView = self.view
                        controller.popoverPresentationController?.sourceRect = CGRect(x: self.view.bounds.midX, y: self.view.bounds.maxY, width: 0, height: 0)
                        controller.popoverPresentationController?.permittedArrowDirections = [.down]
                    }`

philips77, 2019-06-25:
Btw, guys, thank you very much for testing the app. Really appreciate!

PM-BubblyNet, 2019-06-26:
Hey! np! and please let us know if we can help somehow.

closed 2019-06-26 by PM-BubblyNet