nordicsemi/IOS-nRF-Mesh-Library #128: Custom Vendor Messages

d-srd, 2019-06-19:
Hi,

the Android version of the library supports sending custom vendor messages. I've tried implementing the same functionality in the iOS version unsuccessfully.

Here's an overview of the Android functionality I'm trying to recreate:
1. Provision the device
2. Select the settings button of the provisioned device
3. Select the vendor model
4. Bind the app key to the vendor model
5. Scroll to the bottom to send a vendor message
6. Select the 'Acknowledged message' button
7. Enter the opcode and send the message

How could I achieve the same with the iOS version?

I've tried using the 'develop' branch which has a basic implementation of vendor messages. Once I send a message, I don't receive a response from the board I'm using.

The message itself is fairly simple - it doesn't have any arguments passed in, only an opcode.

Feel free to let me know if I'm missing any of the details.

philips77, 2019-06-19:
The currently maintained branch is `verison2`. Vendor messages are not yet supported but they will be soon. I just completed my first successful message exchange yesterday, for Composition Data, so it's not far until Vendor Messages are added. Please, follow the new branch.

Marcello168, 2019-07-20:
@philips77 Continue to work hard，verison2 is better than the version 1

philips77, 2019-08-20:
Vendor Messages should be working now, both using the library and in the sample app. Please, follow this link https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/issues/138#issuecomment-523000792 how to create a vendor message with hardcoded Op Code. The Sample app is just demonstrating the Runtime Vendor Message, where the Op Code is set dynamically.

closed 2019-08-20 by philips77