nordicsemi/IOS-nRF-Mesh-Library #130: About message segmentation

DistanceLe, 2019-06-29:
I found a segmentation bug during use.
For example, when a packet needs to be divided into 10 segments, the mesh device is divided into 0 to 9 segments.
In the previous 0~4 segment, the App can normally receive it normally. then:
1. If the App does not send an Ack packet, the device will repeatedly send the message packet that has been sent before.
2. If the App sends an Ack packet (0~4: 0b 0000 0000 0001 1111), but the device will still send the packets that have been sent, but it is better than not sending the Ack packet.
This means that the device did receive the Ack packet, but the processing logic is problematic.

The logic of my app sending Ack packets is:
1. When the App receives a duplicate message packet, it starts sending Ack packets.
2. Or send an Ack packet when the received segmentation packet is a multiple of 3.

Because the timeout period of a message packet on the device side is 6 seconds. If the segmentation is 10 segments, the app can receive about 20 segments, but many of them are duplicates, even though I have already sent the Ack package.
This causes the device to send a large packet and will often fail.
For example: the message GetComposition, because there are many elements in the node.

philips77, 2019-06-29:
Hi, are you talking about the *master*, or *version2* branch? The new version that's under development is in the other branch.

DistanceLe, 2019-06-29:
Chip model: nRF52832
I mean, your chip's program seems to have bugs in segmentation.

I am wondering if you can confirm with your chip engineer how the chip device handles the segmentation operation.
Because the code of the chip can't be modified here, we can't see it too.

philips77, 2019-06-29:
Which SDK version are you using?

philips77, 2019-06-29:
This is because the connection interval on SDK 3.1 is set to very long 500ms, or something like that. The timeout for each message sent is much shorter, so after sending a packet, it timeouts even before the connected device could reply, causing adding the same packet to the outgoing queue. The queue is not empty when the total ack is received, so it will continue to send data.
It's a known issue, it should be addressed in the next release.

DistanceLe, 2019-07-01:
Ok, when is your new version expected to be released?

Our chip engineers can modify the connection interval. After the connection interval is shortened, the probability of success is indeed greater. But the logic of segmentation still seems to be not optimal, and the number of repeated segmentation packages is still too much.

You can continue to optimize more. We are not able to modify the code of the segmentation logic?

If the connection interval is changed to be shorter, it will cause some other problems？ For example, the life of the chip, or power consumption?

philips77, 2019-07-01:
> Ok, when is your new version expected to be released?

I think Nordic will be anouncing it using official channels. I don't know the schedule of the SDK team.

> Our chip engineers can modify the connection interval. After the connection interval is shortened, the probability of success is indeed greater. But the logic of segmentation still seems to be not optimal, and the number of repeated segmentation packages is still too much.

You have to ask on [DevZone](https://devzone.nordicsemi.com/). This is outside the scope of this library and, to be honest, I have almost zero experience in firmware development. The Support Team is more than qualified to answer such questions, they can also get in contact with the mesh team, if needed.

> If the connection interval is changed to be shorter, it will cause some other problems？ For example, the life of the chip, or power consumption?

For sure yes, you'd have to measure it. I don't know the reason for such long interval in the current SDK, but there must be one.

I'm closing the issue here, as it is not related to the library, but feel free to link to it from DevZone or open another one if you find any other issue.
Also, thank for the feedback!

closed 2019-07-01 by philips77

DistanceLe, 2019-07-01:
OK👌

philips77, 2019-07-01:
Some info from the mesh team: 

> If connection interval is too short, device will miss many mesh packets on ADV bearer.

DistanceLe, 2019-07-01:
Too short: Is there a specific time frame?
The current interval is 500ms, can it be changed shorter?

philips77, 2019-07-01:
I guess you just increase a chance of missing a packet when decreasing connection interval, as the radio is busy sending empty packets to keep the connection open. But of course you may miss some even with 500ms, there is no threshold.

DistanceLe, 2019-07-02:
Okay, I understand.
I look forward to the next version to be done as soon as possible