nordicsemi/IOS-nRF-Mesh-Library #132: Group Setting Subscription

BubblyNetIntegration, 2019-07-17:
Question:

Hey Aleksander I just wanna say welcome back, hope you enjoyed your time off. Quick question. When do you think Group Setting Subscription will be done? We can't wait to start working on it. 

The app is looking very good, congrats!

philips77, 2019-07-18:
I've started working on groups on Tuesday. I still need to polish the API. Next step is adding them to publication and implementing subscription. I guess next week it should all work.
Btw, I'm still on vacation, just coding while enjoying 🌞 ;)

BubblyNetIntegration, 2019-07-18:
@philips77 😅😅😅
Thanks for the quick answer though!! Enjoy the rest of your vacation!

philips77, 2019-07-26:
Hi, subscriptions should be working now. First, create groups, then you may use them in Publications and Subscriptions in Models. I didn't implement Subscription List and Get messages yet, but the rest should work.

closed 2019-07-26 by philips77

BubblyNetIntegration, 2019-07-26:
@philips77 fantastic, thanks for the update!

philips77, 2019-07-29:
Ups, I forgot to push one file that's needed for Subscriptions. I just pushed all the changes, so everything should finally work now.

philips77, 2019-07-29:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/commit/97f24a203c46213da77e33ef8c657449133fccb2