nordicsemi/IOS-nRF-Mesh-Library #133: Create MeshNetwork .noNetworkKey

fiveagle, 2019-07-26:
Hi,
Apparently with the recent added support for Groups the app crashes on first install, when it can't reload the MeshNetwork from disk.

I think the issue has been introduced in MeshNetwork+Provisioner.swift on Commit: Code improvements Jul 23, 2019

**To Reproduce**
Steps to reproduce the behavior:
1. Fresh install
2. Build and Run
3. nRFMeshProvision_Example[1888:1214951] Fatal error: 'try!' expression unexpectedly raised an error: nRFMeshProvision.MeshModelError.noNetworkKey: file /Users/marcodipeco/Downloads/IOS-nRF-Mesh-Library-version2-2/nRFMeshProvision/Classes/MeshNetworkManager.swift, line 104

**Expected behavior**
Create a MeshNetwork and then setup the keys

**Platform details:**
 - Device: iPhone 8 and Simulator
 - OS: 12.2
 - Probably from Commit: Code improvements on Jul 23, 2019

philips77, 2019-07-28:
Yes, adding Groups broke the backwards compatiblity, as they are required in the json. A new mesh network must be generated, or just add an empty "groups": [] in json. Until the lib goes to stable such issues may happen 
But you're getting `noNetworkKey` error, which means it may be something else... I'll have a look at it on Monday.

philips77, 2019-07-28:
I think I know what's the issue. At the beginning, when you create a new mesh network, there's no network key, but a node is created for the provisioner. That causes a crash. I'll fix it on Monday. Thanks for the report!

philips77, 2019-07-29:
Fixed. The key is now mandatory for each network and all nodes.
A default random key is generated with a new network. Instead of giving the used to type the primary network key, there is now an option to modify a not-used key in the sample app.

closed 2019-07-29 by philips77

fiveagle, 2019-07-29:
Great! It works, thank you  very much.