nordicsemi/IOS-nRF-Mesh-Library #136: Bound app keys not shown for Vendor Model with companyIdentifier

nvonahsen, 2019-08-08:
**Describe the bug**
*Using the example application.*
After adding an app key to an element I attempted to navigate to a defined vendor model and bind that app key to it. The bind operation appeared to succeed and the app key index seemed to be returned when looking at the ConfigModelAppList message using breakpoints. However the app key never showed up in the UI for the app. 

After messing around with breakpoints it seems that the companyIdentifer is getting dropped from the modelId in ConfigAnyModelMessage and ConfigVendorModelMessage which means that the library fails to match up the modelId of the incoming message to the model stored in the element (which does have the correct id). 

**Believed cause**
See ConfigMessage lines 245 and 256 (ConfigAnyModelMessage.modelId and ConfigVendorModelMessage.modelId)

Order of conversion to UInt32 and << 16 operator looks wrong. identifier is shifted and since it is only 16 bits it just gets deleted. Then 0 gets converted to UInt32

```swift
    var modelId: UInt32 {
        if let companyIdentifier = companyIdentifier {
            return UInt32(companyIdentifier << 16) | UInt32(modelIdentifier)
        } else {
            return UInt32(modelIdentifier)
        }
    }
    
    var modelId: UInt32 {
        return UInt32(companyIdentifier << 16) | UInt32(modelIdentifier)
    }
```
*Should be*
```swift
    var modelId: UInt32 {
        if let companyIdentifier = companyIdentifier {
            return UInt32(companyIdentifier) << 16 | UInt32(modelIdentifier)
        } else {
            return UInt32(modelIdentifier)
        }
    }
    
    var modelId: UInt32 {
        return UInt32(companyIdentifier) << 16 | UInt32(modelIdentifier)
    }
```

**To Reproduce**
Steps to reproduce the behavior (using the example app):
1. Have a device with a vendor model that includes a company ID (mine is 0x1000)
2. Add an application key to the node
3. Open the vendor model and attempt to bind the application key
4. Application key binds but does not show up on the model page
5. Pull-To-Refreshing the page also doesn't cause it to show up
6. Adding a breakpoint near FoundationLayer line 285 reveals the key list has modelId consisting only of the modelIdentifier (0x59 in my case) while the models in element.models have the combined modelId (0x100059). This then causes no model to be returned from `element.model(withModelId:)`

**Expected behavior**
Model is found correctly based on the returned model/company ID

**Platform details:**
 - Library Commit: c18e6366d01b033ddef1398a86e4c2cb3f5f9000

philips77, 2019-08-08:
Great catch! I'll fix it today. Thanks for testing!

philips77, 2019-08-09:
I've applied your fix, hope it will work now. If not, please reopen the issue. It will be some time before I will be testing Vendor Messages.

closed 2019-08-09 by philips77