nordicsemi/IOS-nRF-Mesh-Library #137: Controlling in Version 2

sunilloharshoreline, 2019-08-09:
Is there any support for controlling any Model either Individual or Group (looks like there is no support from Group :  https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/issues/116  , but is Individual control available)  ?

philips77, 2019-08-09:
Hi, not yet. 
Both individual and group control should be added very soon. Next week or two.

sunilloharshoreline, 2019-08-09:
Thanks for the update.

BubblyNetIntegration, 2019-08-19:
Hello @philips77. I hope you enjoying your vacation. 
Any idea of when the controllers will be implemented?
Thanks!

philips77, 2019-08-19:
I'm implementing controls for Vendor Models right now. Generic On/Off will follow, then those 2 in Groups.

philips77, 2019-08-20:
Vendor Messages can be now sent from the UI. Go to Node -> Element -> Vendor Model -> Controls. Tested using Zephyr implementation.

sunilloharshoreline1, 2019-08-21:
Hi @philips77 , i checked in the latest code pushed to repo, there are compile errors in MeshMessage.swift file for extension CustomDebugStringConvertible. The error says : Extension of protocol 'CustomDebugStringConvertible' cannot have an inheritance clause
Can you look into it ?
Thanks.

philips77, 2019-08-21:
Oh, sure. I'll have a look at it now.

philips77, 2019-08-21:
Done. Sorry, that was stupid. Next time I'll try at least to compile before pushing changes ;)

sunilloharshoreline1, 2019-08-21:
Thanks for the quick fix. I m able to compile successfully but as you mentioned the vendor messages can be sent through UI , i didn't find anything on UI for this. What i did was :  first i provisioned the node and then Assigned App Key and Subscription to the Model (Generic On Off).
Am i missing something ?
Here is the screenshot :
[image]

philips77, 2019-08-21:
Generic ON/Off will be added shortly. Currently you'd see a custom UI at the bottom for Vendor Models.
If you want to test it, here's the Sample On Off app (Client and Server) for nRF52832 DK.
[Mesh SDK 3.2 Simple On Off.zip](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/files/3525075/Mesh.SDK.3.2.Simple.On.Off.zip)
Here you'll find some documentation about the supported messages: https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.meshsdk.v3.2.0/md_models_vendor_simple_on_off_README.html?cp=5_2_2_3_1

philips77, 2019-08-23:
Generic OnOff Server controls should be in place right now if you pull the latest changes.

philips77, 2019-09-06:
I'm closing the issue, as Generic OnOff, Generic Level and Vendor Models can be configured from the UI now. The 2 first can be also controller from the Group level.

closed 2019-09-06 by philips77