nordicsemi/IOS-nRF-Mesh-Library #138: I want to send data to Vendor mode, but I don't know how to combine data.

codepgq, 2019-08-12:
This is code
``` swift
struct VendorMessage: MeshMessage {
    static var opCode: UInt32 = 0xc001
    
    var parameters: Data?
    
    init?(parameters: Data) {
        self.parameters = parameters
    }
    
    init?(vendor: VendorModel) {
        guard let bytes = try? vendor.hex.bytesFromHexString() else {
            return nil
        }
        self.parameters = Data([0x02,0x0c,0x01] + bytes)
    }
}
```

Now I can send it, but nothing happens.
I am not where there are some details that have not been processed。

philips77, 2019-08-12:
Hi,
I'll add support for Vendor Messages soon (hopefully this week). Also, Vendor Messages need to have 3-byte Op Code in format 0b11XXXXXX ZZZZZZZZ ZZZZZZZZ where the Zs are for Company Identifier, and 6-bit Xs is for the Op Code.

philips77, 2019-08-20:
Hi, Vendor Models are now supported in the library and in the sample App. The sample app is using RuntimeVendorMessage (where the Op Code can be set in runtime), but this should not be necessary for "normal" apps, where all opcodes are known and can be hardcoded, as they are in ConfigMessages.

To create your vendor message use the code like this (here the Op Code is 0 and Company ID is 0xABCD):

```swift
struct MyVendorMessage: StaticVendorMessage {
   static let opCode: UInt32 = 0xC0CDAB // ((0xC0 | (6-bit Op Code)) << 16) | CompanyID.bigEndian
    
    var parameters: Data? {
        // Return parameters 
    }

    init?(parameters: Data) {
        // Parse Parameters
    }
}
```
If you are expecting a Status for your Message, register its class in the `MeshNetworkManager` using `register(vendorMessageType:)` so that it can be instantiated. Otherwise you'll receive the `UnknownMessage`.

closed 2019-08-20 by philips77