nordicsemi/IOS-nRF-Mesh-Library #141: Set UnicastAddress for a existing node.

fiveagle, 2019-08-22:
Hi,
One of the requirements of the project I'm working on is to be able to manually set the Unicast Address for a Node.
Right now it looks like is not possible to do it. Is it so because of something particular? Or is going to be available in the future?

Thanks,
Marco Di Peco

philips77, 2019-08-22:
It is possible, also in the Sample app. In the sample app, after connecting to an Unprovisioned device you my modify its Unicast Address and the Network Key to provision it with. When you are using just the library, you have to use this property: https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/efb990d0acea41a7e4fe6502999ff71aaa3e345a/nRFMeshProvision/Classes/Provisioning/ProvisioningManager.swift#L52-L55

When set to `nil`, the address will be assigned automatically based on the next available address for required number of models in your Provisioner's range.

fiveagle, 2019-08-22:
Great! Thanks.

closed 2019-08-22 by philips77