nordicsemi/IOS-nRF-Mesh-Library #143: Device reset problem

DistanceLe, 2019-08-27:
Recently, when I reset the device, there is a very strange phenomenon.
When I sent the reset command, the device also replied, but the device was not actually reset. The configuration information of the device is still there, and the service UUID has not changed.
What do you think will be caused?

But not every time, sometimes it can be reset successfully.

philips77, 2019-08-27:
This is not an issue with the iOS mesh library, rather the sdk. I recommend you asking this question on DevZone, so our Support or Mesh teams could check it out. 
From my side, I'll confirm that we also have seen this issue here with Mesh SDK 3.1, but I think it might have been fixed in 3.2. I'll double check. Also, I think it doesn't work when you're connected directly to this node using gatt, but works when sent via mesh network using adv bearer, but also i can't confirm.

philips77, 2019-08-27:
Hava a look at the second item here: https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.meshsdk.v3.2.0/md_RELEASE_NOTES.html#release_notes_320_bugfixes

closed 2019-08-27 by philips77