nordicsemi/IOS-nRF-Mesh-Library #146: Device not supported after provisioning

fiveagle, 2019-08-28:
Hi,
after I have provisioned the device and I have added it to the MeshNetworkManager connection like in the Sample app, inside the GattBearer gets called the didDisconnectPeripheral with error deviceNotSupported

The device is from Nordic but I got it from Vimar

Steps to reproduce the behavior:
1. Scan like the sample app
2. Start a provision
3. Get the node from the MeshNetwork
4. Save the MeshNetwork

It look like it fails because the .notify characteristic is missing from the dataOutCharaterristic. In the BaseGattProxyBearer class.

**Platform details:**
 - Device: iPhone8
 - OS: 12.x

philips77, 2019-08-28:
What firmware are you using on the device side? Is is Mesh SDK from Nordic? Zephyr? Or some other implementation?
The Mesh Proxy and Mesh Provisioning service should each have 2 characteristics, one with Write Without Response, the other with Notify properties in order to be valid. If you are sure they have them, try restarting Bluetooth on the iPhone to clear the cached services and try again.

fiveagle, 2019-08-28:
I really don't know right now. But after some research I think the problem is that I don't ask the node to EnableNotifications for the ProxyDataOutCharacteristic as my colleagues are doing in Android. I am supposed to use the ConfigGATTProxyStatus and when?

Thanks

philips77, 2019-08-28:
If you're getting `.deviceNotSupported`, it's a problem with your Node's firmware or cache on iOS.
If your using the GattBearer class, it enable notifications automatically.
The ConfigGATTProxy messages can be sent or received only when your Node is working and at up correctly. As written in the documentation, it disables relying messages sent from connected gatt client to mesh network over adv bearer.

What exactly is your issue?

fiveagle, 2019-08-29:
Hi, 
apparently the .deviceNotSupported log gets fired in BaseGattProxyBearer line: 209 but I think is a false warning because it gets inside the guard all the time.

My issue is that I have to bind the appKey on all servers (aka Models) and the publish address on some of them and subscribe on a group address for one other model. But I can't because the model array is empty and I cannot understand what step I am missing after the provisioning.

Anyway thank you for the support.

philips77, 2019-08-29:
You have to send ConfigCompositionDataGet message.

fiveagle, 2019-08-29:
Yes now it works. But I still can't manage to bind the key. I receive some errors when sending the message. It could be that I ask to bind each model too fast. I'm still working on it.
Thank You very much.

philips77, 2019-08-29:
Remember that you have to send the Application Keys before binding them. And perhaps other Network Keys as well.

fiveagle, 2019-08-30:
Yes it goes. But I encounter a lot of device disconnection for .deviceNotSupported during the process. I'm sending the Congif to most of the model one after the other. It reconnects and keeps going most of the time.

philips77, 2019-09-06:
Any update on this?

philips77, 2019-09-27:
I'm closing this issue due to inactivity. Please, open another one if you find any other issue.

closed 2019-09-27 by philips77

fiveagle, 2019-09-27:
Ok thank you anyway. Both me and android managed to make it work if we set a delay of 1.4 s.