nordicsemi/IOS-nRF-Mesh-Library #148: Requesting Composition Data Error

fiveagle, 2019-09-04:
**Describe the bug**
Hi, I am trying to send a ConfigCompositionDataGet but I encounter an error quite often.
The library prints "Error: Segment invalid or repeated" multiple times and then nothing happens neither I receive a bearer delegate message about disconnection. I am using the ConnectableViewController as in the Sample App. 
I have also tried to do a DFU firmware update before the provisioning.

**To Reproduce**
Steps to reproduce the behavior:
1. Provioning a Node
2. I first send a ConfigAppKeyAdd
3. I send the ConfigCompositionDataGet
4. 60% of the times it gets stuck here

**Expected behavior**
To receive some callback for the error

**Platform details:**
 - Device: iPhone 8
 - OS: 12
 - Library Version: last

**Logs / Screenshots**
Requesting Composition Data...
Sending ConfigCompositionDataGet(page: 0) to 0x000B
-> 0x0076AC255794964EF14BEFFF2D05AF90C469837E92C8C7545F
-> 0x007678BA07D6E007B91ABE05BCFCEB33AFC4784420A767C42E
<- 0x0076F44B69F94173126F6D67118E2348D0C15B43324FBC68AB
ACK (000B->0001) for SeqZero: 0 with Block ACK: 0x00000003 received
<- 0x0076BAB51CAC571A9919EDF3215D88D4A0F112DE97C5760114
ACK (000B->0001) for SeqZero: 4 with Block ACK: 0x00000001 received
<- 0x0076EEEA2B6F97AF1A1FFCF05FAE1B403DD541136E79CC0E3FBE7FA96E2C
Segment Segmented Access Message (000B->0001) for SeqZero: 5 (1/3), Seq: 5, 0xAB778198305116B938F280A7 received
Sending ACK (0001->000B) for SeqZero: 5 with Block ACK: 0x00000001
-> 0x007698975EEFFFEA8BF3B9099094E7C42E71D8ACA053032020
<- 0x00761EF78CCD8569A9D26030277A21CC3CA657ED299240EFB0CAF24146E4
Segment Segmented Access Message (000B->0001) for SeqZero: 5 (2/3), Seq: 5, 0xAC13CC497260D057A6CAAEF6 received
Sending ACK (0001->000B) for SeqZero: 5 with Block ACK: 0x00000003
-> 0x0076380E2627C2D310139D750F54E47339B9BF72733C16E440
<- 0x00767AA535B550251A10997B3A472F9E15B0AF36F585E8159B
ACK (000B->0001) for SeqZero: 4 with Block ACK: 0x00000001 received
<- 0x00761E4B7B37DE17517A7F32C3A82FE02D73759509584BF74BD0AC149C65
Segment Segmented Access Message (000B->0001) for SeqZero: 5 (1/3), Seq: 5, 0xAB778198305116B938F280A7 received
Error: Segment invalid or repeated
<- 0x007612583B39DBCDB555CDE7322336EF5F4D4C3B5EBEC92A8B6E341640F6
Segment Segmented Access Message (000B->0001) for SeqZero: 5 (2/3), Seq: 5, 0xAC13CC497260D057A6CAAEF6 received
Error: Segment invalid or repeated
<- 0x0076633F6270D38ADF24DDE5958053DE58596B987F49A43A250240850E5B
Segment Segmented Access Message (000B->0001) for SeqZero: 5 (1/3), Seq: 5, 0xAB778198305116B938F280A7 received
Error: Segment invalid or repeated
<- 0x00765D211D2321341CE5FF5A949732B7A15C405B9E4BCE1A666BA43CEC2C
Segment Segmented Access Message (000B->0001) for SeqZero: 5 (2/3), Seq: 5, 0xAC13CC497260D057A6CAAEF6 received
Error: Segment invalid or repeated

philips77, 2019-09-04:
Do you have any delay between receiving the `ConfigAppKeyStatus` and sending `ConfigCompositionDataGet`? Seems like the device is still sending you ACKs for the previous message.
Also, the `Sending ACK (0001->000B) for SeqZero: 5 with Block ACK: 0x00000003` should be repeated up to 10 times, with the default settings. Did you modify the parameters in `MeshNetworkManager`?
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/ae24651405e74a5db927fb26949b01597bcfa770/nRFMeshProvision/Classes/MeshNetworkManager.swift#L60-L66

fiveagle, 2019-09-04:
I send the ConfigCompositionDataGet 0.7 s after I get the ConfigAppKeyStatus delivered.
I haven't modified the MeshNetworkManager.

I am gonna try with more than 0.7 ms

philips77, 2019-09-06:
Any success?

fiveagle, 2019-09-06:
I heve set a delay between each send of 1.7s. Now it looks like it's more reliable, but I still think it should send some callback to let the app fail gracefully. 
Now I have to check set a timeout to start the process again after some time of inactivity.

philips77, 2019-09-13:
Btw, the acknowledged messages  should be repeated automatically when the status wasn't received. I will implement this next week.

philips77, 2019-09-19:
This may be duplicate of #153. I think I pinpointed the issue and I'm committing a fix very soon.

closed 2019-09-20 by philips77