nordicsemi/IOS-nRF-Mesh-Library #149: About 20 bytes overflowed before IOS 11.

JefferDevs, 2019-09-10:
It's  important. I implemented it, I don't know if it is correct, I am a newbie and I look forward to discussing with you.
```
 // For iOS versions before 11, the data must be just sent in a loop.
 // This may not work if there is more than ~20 packets to be sent, as a
 // buffer may overflow. The solution would be to add some delays, but
 // let's hope it will work as is. For now.
 // TODO: Handle very long packets on iOS 9 and 10.

 for packet in packets {
        let count = packet.count
        let times = Int(ceilf(Float(count) / 20.0))
        for i in 0..<times {
                let pkgLength = (count - i * 20) > 20 ? 20 : count - i * 20
                let start = i * 20
                let end = start + pkgLength
                let pck = packet.subdata(in: start..<end)
                print("-> 0x\(pck.hex)")
                basePeripheral.writeValue(pck, for: dataInCharacteristic, type: .withoutResponse)
         }
  }
```

philips77, 2019-09-11:
I'll check it tomorrow

philips77, 2019-09-12:
Hello,
As written in this comment, iOS versions before 11 didn't have buffer control for sending packets with type `.withoutResponse`. After sending ~20+ packtes in a loop
```swift
for i in 0..<40 {
   basePeripheral.writeValue(pck, for: dataInCharacteristic, type: .withoutResponse)
}
```
the BLE on the iPhone would stack and no further operation would be possible. This is because of buffer overflow. iOS has some outgoing buffer for sending packets and this loop adds new packets much faster then they are actually sent by air, causing eventually a buffer overflow.
On iOS 11+ there are 2 new methods added, one to check if the buffer is available, and a callback when the buffer is ready again, which solve this problem.
For older versions you have to solve is somehow on your own. There are number of options, for example the target peripheral may send a notification every ~10-15 packets to synchronize the sender with receiver, or you may write `.withResponse` every ~10-15th packet, after which you are getting the [peripheral(_ peripheral: CBPeripheral, didWriteValueFor characteristic: CBCharacteristic, error: Error?)](https://developer.apple.com/documentation/corebluetooth/cbperipheraldelegate/1518823-peripheral) callback when the device acknowledges it. You may also add some artificial delay, but it's difficult to done it correctly, as it's easy to wait too short or too long and slow down the transmission.

JefferDevs, 2019-09-17:
I understand, thank you very much.

closed 2019-09-17 by philips77