nordicsemi/IOS-nRF-Mesh-Library #153: Stuck in Requesting Composition Data when trying to provision a 2nd node

icarus31, 2019-09-18:
**Describe the bug**
The application is stuck on 'Requesting Composition Data'

**To Reproduce**
Steps to reproduce the behavior:
1. Provision a node
2. Provision a second node

**Expected behavior**
I expect to work. :-)

**Platform details:**
 - Device: iphone 7, 8, ipad
 - OS: iOS 12.x
 - Library Version (or Commit): 875c712 (but it happened before than that)

**Logs / Screenshots**
There are 2 logs

FIRST LOG

New state: Provisioning Invitation sent
New state: Provisioning Capabilities received:
--- Capabilities ---
Number of elements: 1
Algorithms: FIPS P-256 Elliptic Curve
Public Key Type: None
Static OOB Type: None
Output OOB Size: 0
Output OOB Actions: None
Input OOB Size: 0
Input OOB Actions: None
--------------------
New state: Provisioning started
New state: Provisioning complete
New state: Provisioner is ready
2019-09-18 10:10:33.244508-0400 nRFMeshProvision_Example[3424:1554160] [CoreBluetooth] XPC connection invalid
2019-09-18 10:10:33.244782-0400 nRFMeshProvision_Example[3424:1554160] [CoreBluetooth] XPC connection invalid
2019-09-18 10:10:33.819359-0400 nRFMeshProvision_Example[3424:1554160] [FoundationModel] Sending ConfigCompositionDataGet(page: 0) to: 0003
2019-09-18 10:10:33.822796-0400 nRFMeshProvision_Example[3424:1554160] [Access] Sending Access PDU (opcode: 0x8008, parameters: 0x00)
2019-09-18 10:10:33.824221-0400 nRFMeshProvision_Example[3424:1554160] [UpperTransport] Sending Upper Transport PDU (encrypted data: 0xE52B4C, transMic: 0x3E5E1B00) encrypted using key: Device Key
2019-09-18 10:10:33.826063-0400 nRFMeshProvision_Example[3424:1554160] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 12, segO: 0, segN: 0, data: 0xE52B4C3E5E1B00)
2019-09-18 10:10:33.831621-0400 nRFMeshProvision_Example[3424:1554160] [Network] Sending Network PDU (ivi: 0, nid: 0x4F, ctl: 0, ttl: 5, seq: 12, src: 0001, dst: 0003, transportPdu: 0x08D7A43EB1FA5606CB668E, netMic: 0x70DA67EE) encrypted using Primary Network Key (index: 0)
2019-09-18 10:10:33.833057-0400 nRFMeshProvision_Example[3424:1554160] [Bearer] -> 0x004F1511B93D32AE1B9F08D7A43EB1FA5606CB668E70DA67EE
2019-09-18 10:10:33.916730-0400 nRFMeshProvision_Example[3424:1554131] [Bearer] <- 0x004F5812EBB221210E55300D8B2A614DA33702FD95934AE0C3
2019-09-18 10:10:33.917376-0400 nRFMeshProvision_Example[3424:1554165] [Network] Network PDU (ivi: 0, nid: 0x4F, ctl: 1, ttl: 4, seq: 65536, src: 0003, dst: 0001, transportPdu: 0x300D8B2A614DA3, netMic: 0x3702FD95934AE0C3) received
2019-09-18 10:10:33.918742-0400 nRFMeshProvision_Example[3424:1554165] [LowerTransport] ACK (seqZero: 12, blockAck: 0x00000001) receieved (decrypted using key: Primary Network Key (index: 0))
2019-09-18 10:10:34.007388-0400 nRFMeshProvision_Example[3424:1554131] [Bearer] <- 0x004F3D0714CFFB207A6B5D57E4497221F60B20777C754A59E3FADFA44422
2019-09-18 10:10:34.017788-0400 nRFMeshProvision_Example[3424:1554165] [Network] Network PDU (ivi: 0, nid: 0x4F, ctl: 0, ttl: 4, seq: 65537, src: 0003, dst: 0001, transportPdu: 0x5D57E4497221F60B20777C754A59E3FA, netMic: 0xDFA44422) received
2019-09-18 10:10:34.022000-0400 nRFMeshProvision_Example[3424:1554165] [LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 1, segO: 0, segN: 6, data: 0x42216BBA664E4725A429E0E4) receieved (decrypted using key: Primary Network Key (index: 0))
2019-09-18 10:10:34.066949-0400 nRFMeshProvision_Example[3424:1554131] [Bearer] <- 0x004F37E13F04761699BCA566B4014252964C26DC0FFD24F5FECE804D7CDF
2019-09-18 10:10:34.068196-0400 nRFMeshProvision_Example[3424:1554165] [Network] Network PDU (ivi: 0, nid: 0x4F, ctl: 0, ttl: 4, seq: 65541, src: 0003, dst: 0001, transportPdu: 0xA566B4014252964C26DC0FFD24F5FECE, netMic: 0x804D7CDF) received
2019-09-18 10:10:34.071313-0400 nRFMeshProvision_Example[3424:1554165] [LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 1, segO: 4, segN: 6, data: 0x2562BBB027B34439449FCF34) receieved (decrypted using key: Primary Network Key (index: 0))
2019-09-18 10:10:34.097748-0400 nRFMeshProvision_Example[3424:1554131] [Bearer] <- 0x004F340F00A6A379795BD56A8CD94FA4290EF161BB851CDDC5811DE2F340
2019-09-18 10:10:34.099380-0400 nRFMeshProvision_Example[3424:1554165] [Network] Network PDU (ivi: 0, nid: 0x4F, ctl: 0, ttl: 4, seq: 65542, src: 0003, dst: 0001, transportPdu: 0xD56A8CD94FA4290EF161BB851CDDC581, netMic: 0x1DE2F340) received
2019-09-18 10:10:34.101033-0400 nRFMeshProvision_Example[3424:1554165] [LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 1, segO: 5, segN: 6, data: 0x34707B9B0766EB4BF5C14E5C) receieved (decrypted using key: Primary Network Key (index: 0))
2019-09-18 10:10:34.428438-0400 nRFMeshProvision_Example[3424:1554131] [Bearer] <- 0x004FEF7B72B6627407798577254DF8B4FA8A1AFBD8CDC46DBF7FD00BB9EB
2019-09-18 10:10:34.431152-0400 nRFMeshProvision_Example[3424:1554165] [Network] Network PDU (ivi: 0, nid: 0x4F, ctl: 0, ttl: 4, seq: 65546, src: 0003, dst: 0001, transportPdu: 0x8577254DF8B4FA8A1AFBD8CDC46DBF7F, netMic: 0xD00BB9EB) received
2019-09-18 10:10:34.434724-0400 nRFMeshProvision_Example[3424:1554165] [LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 1, segO: 2, segN: 6, data: 0x81ADCC17FD03A665D37561C5) receieved (decrypted using key: Primary Network Key (index: 0))
2019-09-18 10:10:34.458275-0400 nRFMeshProvision_Example[3424:1554131] [Bearer] <- 0x004F7D2DE66944D5A4A01274ABA03D5BAB21F32B4067ABD2274EC558BFEC
2019-09-18 10:10:34.460606-0400 nRFMeshProvision_Example[3424:1554165] [Network] Network PDU (ivi: 0, nid: 0x4F, ctl: 0, ttl: 4, seq: 65548, src: 0003, dst: 0001, transportPdu: 0x1274ABA03D5BAB21F32B4067ABD2274E, netMic: 0xC558BFEC) received
2019-09-18 10:10:34.462944-0400 nRFMeshProvision_Example[3424:1554165] [LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 1, segO: 4, segN: 6, data: 0x2562BBB027B34439449FCF34) receieved (decrypted using key: Primary Network Key (index: 0))
2019-09-18 10:10:34.463065-0400 nRFMeshProvision_Example[3424:1554165] [LowerTransport] Segment invalid or repeated
2019-09-18 10:10:34.488011-0400 nRFMeshProvision_Example[3424:1554131] [Bearer] <- 0x004F9CCEDA3B189E618E31A4001462CA19044A66DFA61EFA5AE1E8A7AEC1
2019-09-18 10:10:34.490215-0400 nRFMeshProvision_Example[3424:1554165] [Network] Network PDU (ivi: 0, nid: 0x4F, ctl: 0, ttl: 4, seq: 65549, src: 0003, dst: 0001, transportPdu: 0x31A4001462CA19044A66DFA61EFA5AE1, netMic: 0xE8A7AEC1) received
2019-09-18 10:10:34.493839-0400 nRFMeshProvision_Example[3424:1554165] [LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 1, segO: 5, segN: 6, data: 0x34707B9B0766EB4BF5C14E5C) receieved (decrypted using key: Primary Network Key (index: 0))
2019-09-18 10:10:34.494049-0400 nRFMeshProvision_Example[3424:1554165] [LowerTransport] Segment invalid or repeated
2019-09-18 10:10:34.519023-0400 nRFMeshProvision_Example[3424:1554131] [Bearer] <- 0x004F60E32512E29D007901226F9A4F20AA3F767921EB3B592684
2019-09-18 10:10:34.521453-0400 nRFMeshProvision_Example[3424:1554165] [Network] Network PDU (ivi: 0, nid: 0x4F, ctl: 0, ttl: 4, seq: 65550, src: 0003, dst: 0001, transportPdu: 0x01226F9A4F20AA3F767921EB, netMic: 0x3B592684) received
2019-09-18 10:10:34.524818-0400 nRFMeshProvision_Example[3424:1554165] [LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 1, segO: 6, segN: 6, data: 0x7C76FEB32CEC9C61) receieved (decrypted using key: Primary Network Key (index: 0))
[... 56 lines cut ...]
2019-09-18 07:58:42.472694-0400 nRFMeshProvision_Example[3366:1532915] [Network] Network PDU (ivi: 0, nid: 0x6A, ctl: 0, ttl: 4, seq: 65538, src: 0004, dst: 0001, transportPdu: 0x65F70C0DB4AB38E0D8E35251D1940A89, netMic: 0x4C509AA9) received
2019-09-18 07:58:42.474388-0400 nRFMeshProvision_Example[3366:1532915] [LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 1, segO: 1, segN: 6, data: 0x1108F8D4E5FE54F3BDFFAE98) receieved (decrypted using key: Primary Network Key (index: 0))
2019-09-18 07:58:42.474475-0400 nRFMeshProvision_Example[3366:1532915] [LowerTransport] Segment invalid or repeated
2019-09-18 07:58:42.502973-0400 nRFMeshProvision_Example[3366:1531603] [Bearer] <- 0x006AAF5E436882222E0B55ED5B0CD882850704379C17C4DDE73E04B939D3
2019-09-18 07:58:42.504448-0400 nRFMeshProvision_Example[3366:1532915] [Network] Network PDU (ivi: 0, nid: 0x6A, ctl: 0, ttl: 4, seq: 65541, src: 0004, dst: 0001, transportPdu: 0x55ED5B0CD882850704379C17C4DDE73E, netMic: 0x04B939D3) received
[... cut ...]
2019-09-18 07:58:42.566824-0400 nRFMeshProvision_Example[3366:1532915] [LowerTransport] Sending ACK (seqZero: 1, blockAck: 0x0000007F)
2019-09-18 07:58:42.569509-0400 nRFMeshProvision_Example[3366:1532915] [Network] Sending Network PDU (ivi: 0, nid: 0x6A, ctl: 1, ttl: 5, seq: 1446, src: 0001, dst: 0004, transportPdu: 0xF237BC9B6F05D6, netMic: 0x0D1A8BE96D744885) encrypted using Primary Network Key (index: 0)
2019-09-18 07:58:42.570085-0400 nRFMeshProvision_Example[3366:1532915] [Bearer] -> 0x006A3BF85EF4FBA17FD6F237BC9B6F05D60D1A8BE96D744885

icarus31, 2019-09-18:
After few investigation, I found that the changes that I did in ConigurationViewController.swift seems to affect the behavior.

in the MeshNetworkDelegate section, I added 2 new cases to the switch:

       ```
 // Handle the message based on its type.
        switch message {
            
        case is ConfigCompositionDataStatus:
            tableView.reloadData()
//            getTtl()
            setGATTProxy(true)
            
        case is ConfigGATTProxyStatus:
            setRelay(count: 1, steps: 1)
            
        case is ConfigRelayStatus:
            setTtl(127)
        
        case is ConfigDefaultTtlStatus:
            done() {
                self.tableView.reloadRows(at: [.ttl], with: .automatic)
                self.refreshControl?.endRefreshing()
            }
            
        case is ConfigNodeResetStatus:
            done() {
                self.navigationController?.popViewController(animated: true)
            }
            
        default:
            break
        }
```

Do you see what could be the issue by adding that?

philips77, 2019-09-18:
I'm working now on repeating acknowledged messages if the response isn't received in some time. Perhaps this will help. Looks like the sdk needs some time before handing messages. I experience similar behavior with the original app (without your changes), the Composition Data are received, the TTL request is acknowledged on lower transport later, but the status message doesn't come. Repeating request should help, I think. You may also try adding some delays and see if it helped.
I'll test this properly before releasing version 2.

philips77, 2019-09-18:
Also, why node 0004 replied with ack worth seq 22v and then 65536 again?

philips77, 2019-09-18:
I'll add a log message when the timeout occurs. Seems like the lib just stopped sending ACKs for some reason.

icarus31, 2019-09-18:
Regarding adding a delay between each set (proxy, relay, ttl) did not work.

I also added a .sync() to the userdefault when incrementing the sequence number in case it was taking to much time to save, but did not work also.

```
defaults.set(sequence + 1, forKey: "S\(pdu.source.hex)")
defaults.sync()
```

I even tried to put the 3 tasks within a dispatchqueue, but did not help.

Always getting

> Segment invalid or repeated

Regarding the sequence number, don't forget both logs were done in different time and maybe with different devices. Still, I will try to take an attention next time.

philips77, 2019-09-19:
But the 22 and 65536 seq was in the second log alone, around 5th line.

philips77, 2019-09-19:
Hi, please give it a spin now.
This commit 13fa5f20f3bb5b7f16cf48b1af7d1ee1abe7e514 caused the lower level timers to stop working, as the new queue didn't have a looper. None of polling ACKs, timeouts, etc were actually triggered anymore. 
Now, with the new `BackgroundTimer` it should work again.

philips77, 2019-09-19:
One other thing that's not yet implemented, is the callback for acknowledged messages. E.g., if the `ConfigCompositionDataGet` message will be sent successfully, but the `ConfigCompositionDataStatus` will not be received before the timeout occurs, the app will not get any callback. I'm implementing a fix for this now.

icarus31, 2019-09-19:
Just to let you know that the timeout got triggered now like you can see in part of my log

```
2019-09-19 07:06:08.504084-0400 nRFMeshProvision_Example[3770:1646181] [LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 1, segO: 5, segN: 6, data: 0x41898606F63E3F87F25ACD7D) receieved (decrypted using key: Primary Network Key (index: 0))
2019-09-19 07:06:09.280931-0400 nRFMeshProvision_Example[3770:1646151] [LowerTransport] Sending ACK (seqZero: 1, blockAck: 0x00000077)
2019-09-19 07:06:09.283634-0400 nRFMeshProvision_Example[3770:1646151] [Network] Sending Network PDU (ivi: 0, nid: 0x05, ctl: 1, ttl: 5, seq: 12, src: 0001, dst: 0003, transportPdu: 0x98DF1FE53370DA, netMic: 0x06F9D06B5694C97F) encrypted using Primary Network Key (index: 0)
2019-09-19 07:06:09.284490-0400 nRFMeshProvision_Example[3770:1646151] [Bearer] -> 0x00057CF1A6C6068A851C98DF1FE53370DA06F9D06B5694C97F
2019-09-19 07:06:18.509536-0400 nRFMeshProvision_Example[3770:1646181] [LowerTransport] Incomplete message timeout: cancelling message (src: 0003, seqZero: 1)
```

Most of the time, refreshing the view do the work properly.

philips77, 2019-09-19:
Yes, the timeout is now triggered correctly (only on logs), but the app still displays "Reading Composition Data..." as it didn't receive any callback.
When I finish acknowledged messages, the Access Layer will resend the request several times, or at least notify you that the expected status hasn't been received.
But my Mac decided to upgrade now, so... this may take some time.

philips77, 2019-09-20:
Resending and timeout for acknowledged messages is now complete and pushed. You may set the interval and timeout parameters through the `MeshNetworkManager`.

closed 2019-09-20 by philips77

PhilippeRivard, 2020-02-20:
The example app and app store app still has this issue. Why is this bug "Closed"?

philips77, 2020-02-20:
The timeout depends on the network settings, number of nodes, size of composition data, etc.
In the current version of the app, on Proxy tab, you'll find a switch to enable manual connection. Turn it on and connect to your newly provisioned device. Then open Network tab and open the device page to obtain Composition Data directly from this node using GATT Proxy protocol.