nordicsemi/IOS-nRF-Mesh-Library #154: Requesting Data Comp fails very often

BubblyNetDev, 2019-09-18:
**Describe the bug**

In the latest version2 sample app getting composition data fails very often. Retrying fails, sometimes restarting the app and navigating to the node again works but not often.

**To Reproduce**
Steps to reproduce the behavior:
1. Provision Node
2. Get Comp Data

**Expected behavior**
Data Composition is successfully received. 

**Platform details:**
 - Device: iPhone 7 Plus
 - OS: iOS 12.4.1
 - Library Version latest

**Logs / Screenshots**

New state: Provisioner is ready
2019-09-18 18:03:11.171136-0400 nRFMeshProvision_Example[787:87377] [Bearer] Central Manager state changed to .poweredOn

2019-09-18 18:03:11.172655-0400 nRFMeshProvision_Example[787:87377] [Bearer] Connecting to v3.2.3 Mesh Driver...

2019-09-18 18:03:11.464320-0400 nRFMeshProvision_Example[787:87392] [CoreBluetooth] XPC connection invalid

2019-09-18 18:03:11.469239-0400 nRFMeshProvision_Example[787:87392] [CoreBluetooth] XPC connection invalid

2019-09-18 18:03:12.041807-0400 nRFMeshProvision_Example[787:87392] [FoundationModel] Sending ConfigCompositionDataGet(page: 0) to: 0002

2019-09-18 18:03:12.042672-0400 nRFMeshProvision_Example[787:87392] [Access] Sending Access PDU (opcode: 0x8008, parameters: 0x00)

2019-09-18 18:03:12.043660-0400 nRFMeshProvision_Example[787:87392] [UpperTransport] Sending Upper Transport PDU (encrypted data: 0x2BFB94, transMic: 0x96F7668B) encrypted using key: Device Key

2019-09-18 18:03:12.044726-0400 nRFMeshProvision_Example[787:87392] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 28, segO: 0, segN: 0, data: 0x2BFB9496F7668B)

2019-09-18 18:03:12.047465-0400 nRFMeshProvision_Example[787:87392] [Network] Sending Network PDU (ivi: 0, nid: 0x3A, ctl: 0, ttl: 5, seq: 28, src: 0001, dst: 0002, transportPdu: 0xD9427549299F708BECE84C, netMic: 0x850AAEE9) encrypted using Primary Network Key (index: 0)

2019-09-18 18:03:13.071878-0400 nRFMeshProvision_Example[787:87377] [Bearer] Connected to v3.2.3 Mesh Driver

2019-09-18 18:03:13.072071-0400 nRFMeshProvision_Example[787:87377] [Bearer] Discovering services...

2019-09-18 18:03:13.886815-0400 nRFMeshProvision_Example[787:87377] [Bearer] Service found

2019-09-18 18:03:13.887005-0400 nRFMeshProvision_Example[787:87377] [Bearer] Discovering characteristrics...

2019-09-18 18:03:14.877400-0400 nRFMeshProvision_Example[787:87377] [Bearer] Data In characteristic found

2019-09-18 18:03:14.877613-0400 nRFMeshProvision_Example[787:87377] [Bearer] Data Out characteristic found

2019-09-18 18:03:14.877788-0400 nRFMeshProvision_Example[787:87377] [Bearer] Enabling notifications...

2019-09-18 18:03:16.360985-0400 nRFMeshProvision_Example[787:87377] [Bearer] Data Out notifications enabled

2019-09-18 18:03:16.361170-0400 nRFMeshProvision_Example[787:87377] [Bearer] GATT Bearer open and ready

2019-09-18 18:03:16.362539-0400 nRFMeshProvision_Example[787:87377] [Bearer] -> 0x003A529C49D510904232D9427549299F708BECE84C850AAEE9

2019-09-18 18:03:16.364510-0400 nRFMeshProvision_Example[787:87377] [Bearer] <- 0x01010095CFAFC1DF552E2A00000000F6F1975A2C749486

2019-09-18 18:03:16.365478-0400 nRFMeshProvision_Example[787:87392] [Network] PDU already handled

2019-09-18 18:03:17.352895-0400 nRFMeshProvision_Example[787:87377] [Bearer] <- 0x003A56CEA8DA64247DA638F1187FD7B51CCAC4E1487445B373

2019-09-18 18:03:17.355226-0400 nRFMeshProvision_Example[787:87445] [Network] Network PDU (ivi: 0, nid: 0x3A, ctl: 1, ttl: 8, seq: 0, src: 0002, dst: 0001, transportPdu: 0x38F1187FD7B51C, netMic: 0xCAC4E1487445B373) received

2019-09-18 18:03:17.359931-0400 nRFMeshProvision_Example[787:87445] [LowerTransport] ACK (seqZero: 28, blockAck: 0x00000001) receieved (decrypted using key: Primary Network Key (index: 0))

2019-09-18 18:03:17.847808-0400 nRFMeshProvision_Example[787:87377] [Bearer] <- 0x003A1111AACCA49BF168451CD646B60D113EBC31ACA383A63A6240B11828

2019-09-18 18:03:17.850130-0400 nRFMeshProvision_Example[787:87458] [Network] Network PDU (ivi: 0, nid: 0x3A, ctl: 0, ttl: 30, seq: 1, src: 0002, dst: 0001, transportPdu: 0x451CD646B60D113EBC31ACA383A63A62, netMic: 0x40B11828) received

2019-09-18 18:03:17.853909-0400 nRFMeshProvision_Example[787:87458] [LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 1, segO: 0, segN: 5, data: 0xF98441CB72A5A36F71FCE8BF) receieved (decrypted using key: Primary Network Key (index: 0))

2019-09-18 18:03:18.343134-0400 nRFMeshProvision_Example[787:87377] [Bearer] <- 0x003AFDCFD79368A622E4CC0CDF457822451CB37C49B08002FCCF295B8590

2019-09-18 18:03:18.345576-0400 nRFMeshProvision_Example[787:87445] [Network] Network PDU (ivi: 0, nid: 0x3A, ctl: 0, ttl: 30, seq: 2, src: 0002, dst: 0001, transportPdu: 0xCC0CDF457822451CB37C49B08002FCCF, netMic: 0x295B8590) received

2019-09-18 18:03:18.349486-0400 nRFMeshProvision_Example[787:87445] [LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 1, segO: 1, segN: 5, data: 0xA0CADA21E802877490AB53B4) receieved (decrypted using key: Primary Network Key (index: 0))

2019-09-18 18:03:18.837249-0400 nRFMeshProvision_Example[787:87377] [Bearer] <- 0x003A838A9C492E05A1D5161252427B54E9875A711EEF12625E8EB274A84D

2019-09-18 18:03:18.839430-0400 nRFMeshProvision_Example[787:87445] [Network] Network PDU (ivi: 0, nid: 0x3A, ctl: 0, ttl: 30, seq: 3, src: 0002, dst: 0001, transportPdu: 0x161252427B54E9875A711EEF12625E8E, netMic: 0xB274A84D) received

2019-09-18 18:03:18.842768-0400 nRFMeshProvision_Example[787:87445] [LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 1, segO: 2, segN: 5, data: 0x5433D1B18721F95B2086FA6F) receieved (decrypted using key: Primary Network Key (index: 0))

2019-09-18 18:03:19.333076-0400 nRFMeshProvision_Example[787:87377] [Bearer] <- 0x003A54811E8EB0812DE8AB39D601419FEB802E072E485C2E8AF92F81B564

2019-09-18 18:03:19.335460-0400 nRFMeshProvision_Example[787:87445] [Network] Network PDU (ivi: 0, nid: 0x3A, ctl: 0, ttl: 30, seq: 4, src: 0002, dst: 0001, transportPdu: 0xAB39D601419FEB802E072E485C2E8AF9, netMic: 0x2F81B564) received

2019-09-18 18:03:19.338942-0400 nRFMeshProvision_Example[787:87445] [LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 1, segO: 3, segN: 5, data: 0x7206F8B035610ED25922E68B) receieved (decrypted using key: Primary Network Key (index: 0))

2019-09-18 18:03:19.827851-0400 nRFMeshProvision_Example[787:87377] [Bearer] <- 0x003A6F40B1EDA4BFCD053D919F539FDBC6A99A02DF6C9CD3A7A54832B4BD

2019-09-18 18:03:19.830084-0400 nRFMeshProvision_Example[787:87445] [Network] Network PDU (ivi: 0, nid: 0x3A, ctl: 0, ttl: 30, seq: 7, src: 0002, dst: 0001, transportPdu: 0x3D919F539FDBC6A99A02DF6C9CD3A7A5, netMic: 0x4832B4BD) received

2019-09-18 18:03:19.833474-0400 nRFMeshProvision_Example[787:87445] [LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 1, segO: 0, segN: 5, data: 0xF98441CB72A5A36F71FCE8BF) receieved (decrypted using key: Primary Network Key (index: 0))

2019-09-18 18:03:19.833736-0400 nRFMeshProvision_Example[787:87445] [LowerTransport] Segment invalid or repeated

philips77, 2019-09-19:
Thank you for the report. I'll look into it today.

philips77, 2019-09-19:
The issue has been fixed today. Please, try it now. Also, have a look at #153.

BubblyNetDev, 2019-09-19:
Just tried it seems to be working now, thanks!

closed 2019-09-19 by BubblyNetDev