nordicsemi/IOS-nRF-Mesh-Library #155: Is it possible to provide brightness control of the lamp?

huahuali, 2019-09-19:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

philips77, 2019-09-19:
The Generic Level messages are implemented. Then it's up to your lamp to support this Model.

closed 2019-09-19 by philips77

huahuali, 2019-09-23:
Can you tell us more about it? Combined with generic level server

huahuali, 2019-09-23:
Or what interface is called, what parameters are passed. Thanks.

philips77, 2019-09-23:
For Generic Level Server model, please read chapters 3.3.2 and 3.2.2 in *Mesh Model Specification 1.0.1* on this page: https://www.bluetooth.com/specifications/mesh-specifications/
Regarding how to send the message to this server, have a look at this line: 
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/cc78cba50f6c0dba80c2d8912981b10a5e9143c1/Example/nRFMeshProvision/View%20Controllers/Network/Configuration/Model/GenericLevelViewCell.swift#L217-L222

The full documentation how to use this library should be online this or next week, when I finish implementing the second version of the mesh library.

huahuali, 2019-09-23:
What you mean is to encapsulate the model and design the page to implement the brightness according to the level.
let me try.

I hope that you also have an operation about brightness as soon as possible. Thanks

philips77, 2019-09-23:
What do you mean by "operation about brightness"? The Generic Level Server model should be used to control brightness, and it is already implemented.

huahuali, 2019-09-23:
I want to use slider to control the brightness of the light, whether it needs to package Light Lightness Get (0x82 0x4B) or other. What else do you need to do?

philips77, 2019-09-23:
Ok, I understand. So far we've implemented Generic Level model. For brightness control, the Light Lightness messages should be better, as human lightness perception isn't linear. 
We're now closing the version 2.
I'll try to add as many light messages as possible before that.