nordicsemi/IOS-nRF-Mesh-Library #156: [CoreBluetooth] XPC connection invalid on Connection

sunilloharshoreline1, 2019-09-19:
Getting an : [CoreBluetooth] XPC connection invalid, if removed action on alert for the Cancel button from ScannerTableViewController.
The code I removed :

alert!.addAction(UIAlertAction(title: "Cancel", style: .cancel) { action in
            action.isEnabled = false
            self.alert!.title   = "Aborting"
            self.alert!.message = "Cancelling connection..."
            bearer.close()
        })

or even though the line **bearer.close()** is commented or removed then the issue gets reproduced

philips77, 2019-10-30:
Hi, did you manage to solve your issue?

sunilloharshoreline1, 2019-10-31:
no luck so far. still getting the same error.

philips77, 2019-11-05:
I think this error happens when the app looses reference to a `CBPeripheral` that is still not disconnected. I see in the code that we start disconnecting and immediately then forget the reference, which seems to generate this warning. Still, this is not a big issue, as the system can clean up on its own.

philips77, 2020-04-29:
The log happens as the `CBCentralManager` is not held by a view controller, or that view controller is being released. When a strong reference to it is lost this message is printed. I think Apple wants us to use just one central manager per app? I don't think anything can be done with it. It doesn't break anything. This log can just be ignored.

In the nRF Mesh Provision library each bearer object holds its own central manager inside. Closing a bearer and loosing a reference to it causes this issue. Also, scanner for unprovisioned devices and for proxy nodes also use new instances of central manager and they drop it when the bearer is connected. Still no issue.

closed 2020-04-29 by philips77