nordicsemi/IOS-nRF-Mesh-Library #157: Request Time Out when configuring app key

icarus31, 2019-09-19:
**Describe the bug**
When trying to apply an app key to a second node, I'm getting a 'Request timed out' error.

**To Reproduce**
Steps to reproduce the behavior:
1. Provision a node
2. Provision a second noe
3. Go to 'Add Application Key'
4. Tap 'Done'

**Expected behavior**
It should add the app key

**Platform details:**
 - Device: all
 - OS: iOS12.x
 - Library Version (or Commit): f96fc3b

**Logs / Screenshots**

```
2019-09-19 09:51:24.103201-0400 nRFMeshProvision_Example[3799:1666100] [FoundationModel] Sending ConfigAppKeyAdd(networkKeyIndex: 0, applicationKeyIndex: 0, key: 16 bytes) to: 0002
2019-09-19 09:51:24.104128-0400 nRFMeshProvision_Example[3799:1666100] [Access] Sending Access PDU (opcode: 0x 0, parameters: 0x00000088043C3ECF429E2B552C862174601F29)
2019-09-19 09:51:24.105194-0400 nRFMeshProvision_Example[3799:1666100] [UpperTransport] Sending Upper Transport PDU (encrypted data: 0x92A12F54034D1335B18F598C9FBB9074BCADCB2F, transMic: 0xB34D0C2E) encrypted using key: 33:E4's Device Key
2019-09-19 09:51:24.106305-0400 nRFMeshProvision_Example[3799:1666100] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 396, segO: 0, segN: 1, data: 0x92A12F54034D1335B18F598C)
2019-09-19 09:51:24.108911-0400 nRFMeshProvision_Example[3799:1666100] [Network] Sending Network PDU (ivi: 0, nid: 0x23, ctl: 0, ttl: 5, seq: 396, src: 0001, dst: 0002, transportPdu: 0x23A20D588384881B4D64F9AB0E4EC1CC, netMic: 0x0ADB3F5B) encrypted using Primary Network Key (index: 0)
2019-09-19 09:51:24.110217-0400 nRFMeshProvision_Example[3799:1666100] [Bearer] -> 0x0023DDC38F6C2DA137AF23A20D588384881B4D64F9AB0E4EC1CC0ADB3F5B
2019-09-19 09:51:24.111256-0400 nRFMeshProvision_Example[3799:1666100] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 396, segO: 1, segN: 1, data: 0x9FBB9074BCADCB2FB34D0C2E)
2019-09-19 09:51:24.118132-0400 nRFMeshProvision_Example[3799:1666100] [Network] Sending Network PDU (ivi: 0, nid: 0x23, ctl: 0, ttl: 5, seq: 397, src: 0001, dst: 0002, transportPdu: 0x10CBD1171DF7F3FEF09D4098F61DF3D7, netMic: 0xC32C68DD) encrypted using Primary Network Key (index: 0)
2019-09-19 09:51:24.119451-0400 nRFMeshProvision_Example[3799:1666100] [Bearer] -> 0x0023BD02A13766F9DAC910CBD1171DF7F3FEF09D4098F61DF3D7C32C68DD
2019-09-19 09:51:25.051084-0400 nRFMeshProvision_Example[3799:1666100] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 396, segO: 0, segN: 1, data: 0x92A12F54034D1335B18F598C)
2019-09-19 09:51:25.057025-0400 nRFMeshProvision_Example[3799:1666100] [Network] Sending Network PDU (ivi: 0, nid: 0x23, ctl: 0, ttl: 5, seq: 398, src: 0001, dst: 0002, transportPdu: 0x7A28B9254264AB4FD57ACFC198A93C7F, netMic: 0xDC1D4475) encrypted using Primary Network Key (index: 0)
2019-09-19 09:51:25.058758-0400 nRFMeshProvision_Example[3799:1666100] [Bearer] -> 0x00232D01F2A651B084D47A28B9254264AB4FD57ACFC198A93C7FDC1D4475
2019-09-19 09:51:25.062200-0400 nRFMeshProvision_Example[3799:1666100] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 396, segO: 1, segN: 1, data: 0x9FBB9074BCADCB2FB34D0C2E)
2019-09-19 09:51:25.064859-0400 nRFMeshProvision_Example[3799:1666100] [Network] Sending Network PDU (ivi: 0, nid: 0x23, ctl: 0, ttl: 5, seq: 399, src: 0001, dst: 0002, transportPdu: 0xE9C5803ABCC864017EB3665ADE532486, netMic: 0x1FEB7D4F) encrypted using Primary Network Key (index: 0)
2019-09-19 09:51:25.065745-0400 nRFMeshProvision_Example[3799:1666100] [Bearer] -> 0x00236C253BEF5DBC2886E9C5803ABCC864017EB3665ADE5324861FEB7D4F
2019-09-19 09:51:25.968354-0400 nRFMeshProvision_Example[3799:1666100] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 396, segO: 0, segN: 1, data: 0x92A12F54034D1335B18F598C)
2019-09-19 09:51:25.973037-0400 nRFMeshProvision_Example[3799:1666100] [Network] Sending Network PDU (ivi: 0, nid: 0x23, ctl: 0, ttl: 5, seq: 400, src: 0001, dst: 0002, transportPdu: 0x93124DB188D468A5730CC9C07E5ED961, netMic: 0x29F08ACF) encrypted using Primary Network Key (index: 0)
2019-09-19 09:51:25.974652-0400 nRFMeshProvision_Example[3799:1666100] [Bearer] -> 0x00232F9B25144B6525B793124DB188D468A5730CC9C07E5ED96129F08ACF
2019-09-19 09:51:25.977977-0400 nRFMeshProvision_Example[3799:1666100] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 396, segO: 1, segN: 1, data: 0x9FBB9074BCADCB2FB34D0C2E)
2019-09-19 09:51:25.998376-0400 nRFMeshProvision_Example[3799:1666100] [Network] Sending Network PDU (ivi: 0, nid: 0x23, ctl: 0, ttl: 5, seq: 401, src: 0001, dst: 0002, transportPdu: 0x58CFD386E47364197843DDBCF4DEC84C, netMic: 0x7E0B687A) encrypted using Primary Network Key (index: 0)
2019-09-19 09:51:26.000887-0400 nRFMeshProvision_Example[3799:1666100] [Bearer] -> 0x0023AA5ECBDEBBEDDB9558CFD386E47364197843DDBCF4DEC84C7E0B687A
2019-09-19 09:51:26.919831-0400 nRFMeshProvision_Example[3799:1666100] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 396, segO: 0, segN: 1, data: 0x92A12F54034D1335B18F598C)
2019-09-19 09:51:26.922742-0400 nRFMeshProvision_Example[3799:1666100] [Network] Sending Network PDU (ivi: 0, nid: 0x23, ctl: 0, ttl: 5, seq: 402, src: 0001, dst: 0002, transportPdu: 0x72192868D17D03433F440E81B4F744B7, netMic: 0xE12425A6) encrypted using Primary Network Key (index: 0)
2019-09-19 09:51:26.924004-0400 nRFMeshProvision_Example[3799:1666100] [Bearer] -> 0x00236A2397A86A31FFBC72192868D17D03433F440E81B4F744B7E12425A6
2019-09-19 09:51:26.925476-0400 nRFMeshProvision_Example[3799:1666100] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 396, segO: 1, segN: 1, data: 0x9FBB9074BCADCB2FB34D0C2E)
2019-09-19 09:51:26.928699-0400 nRFMeshProvision_Example[3799:1666100] [Network] Sending Network PDU (ivi: 0, nid: 0x23, ctl: 0, ttl: 5, seq: 403, src: 0001, dst: 0002, transportPdu: 0xFA4057ED42F8B01219ED6854D3DFB1E2, netMic: 0x8D9F7F1E) encrypted using Primary Network Key (index: 0)
2019-09-19 09:51:26.929595-0400 nRFMeshProvision_Example[3799:1666100] [Bearer] -> 0x00230B2296A20FDFF04BFA4057ED42F8B01219ED6854D3DFB1E28D9F7F1E
2019-09-19 09:51:27.783430-0400 nRFMeshProvision_Example[3799:1666100] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 396, segO: 0, segN: 1, data: 0x92A12F54034D1335B18F598C)
2019-09-19 09:51:27.788080-0400 nRFMeshProvision_Example[3799:1666100] [Network] Sending Network PDU (ivi: 0, nid: 0x23, ctl: 0, ttl: 5, seq: 404, src: 0001, dst: 0002, transportPdu: 0xF81FD50C25F4BDF6B0FEC5A8CB595634, netMic: 0xA7987D6C) encrypted using Primary Network Key (index: 0)
2019-09-19 09:51:27.789695-0400 nRFMeshProvision_Example[3799:1666100] [Bearer] -> 0x0023BD45D18323FF36CBF81FD50C25F4BDF6B0FEC5A8CB595634A7987D6C
2019-09-19 09:51:27.792878-0400 nRFMeshProvision_Example[3799:1666100] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 396, segO: 1, segN: 1, data: 0x9FBB9074BCADCB2FB34D0C2E)
2019-09-19 09:51:27.795269-0400 nRFMeshProvision_Example[3799:1666100] [Network] Sending Network PDU (ivi: 0, nid: 0x23, ctl: 0, ttl: 5, seq: 405, src: 0001, dst: 0002, transportPdu: 0x599894466323B87A973F28D63FF0A5F4, netMic: 0x77C68704) encrypted using Primary Network Key (index: 0)
2019-09-19 09:51:27.796158-0400 nRFMeshProvision_Example[3799:1666100] [Bearer] -> 0x0023FD48D2379343E59C599894466323B87A973F28D63FF0A5F477C68704
2019-09-19 09:51:28.698766-0400 nRFMeshProvision_Example[3799:1666100] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 396, segO: 0, segN: 1, data: 0x92A12F54034D1335B18F598C)
2019-09-19 09:51:28.703830-0400 nRFMeshProvision_Example[3799:1666100] [Network] Sending Network PDU (ivi: 0, nid: 0x23, ctl: 0, ttl: 5, seq: 406, src: 0001, dst: 0002, transportPdu: 0xBA89D575D20019CFC316757AC2A506F7, netMic: 0x1F9F4081) encrypted using Primary Network Key (index: 0)
2019-09-19 09:51:28.705478-0400 nRFMeshProvision_Example[3799:1666100] [Bearer] -> 0x00239E89C21AD2FA229EBA89D575D20019CFC316757AC2A506F71F9F4081
2019-09-19 09:51:28.708863-0400 nRFMeshProvision_Example[3799:1666100] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 396, segO: 1, segN: 1, data: 0x9FBB9074BCADCB2FB34D0C2E)
2019-09-19 09:51:28.713022-0400 nRFMeshProvision_Example[3799:1666100] [Network] Sending Network PDU (ivi: 0, nid: 0x23, ctl: 0, ttl: 5, seq: 407, src: 0001, dst: 0002, transportPdu: 0x2C119511EB31CEE00E161BCF26987814, netMic: 0x3F978CD5) encrypted using Primary  [... cut ...]
2019-09-19 09:51:33.125153-0400 nRFMeshProvision_Example[3799:1666154] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 396, segO: 1, segN: 1, data: 0x9FBB9074BCADCB2FB34D0C2E)
2019-09-19 09:51:33.128254-0400 nRFMeshProvision_Example[3799:1666154] [Network] Sending Network PDU (ivi: 0, nid: 0x23, ctl: 0, ttl: 5, seq: 417, src: 0001, dst: 0002, transportPdu: 0x1B3841E4EF71116FA6D83272AFEB92A4, netMic: 0xFD66E85E) encrypted using Primary Network Key (index: 0)
2019-09-19 09:51:33.129174-0400 nRFMeshProvision_Example[3799:1666154] [Bearer] -> 0x0023C38F0EC1160542851B3841E4EF71116FA6D83272AFEB92A4FD66E85E
```

philips77, 2019-09-20:
Hi,
I don't see anything wrong here. Seems like the node 0002 either cannot decode the message, doesn't exist, is out of range. Also, the node you are connected to may have the GATT Proxy feature disabled. The app tried 11 times and at the end it thrown timeout, as no ACK was received.

icarus31, 2019-09-20:
Hi,
Can't reproduce it like yesterday.

Let's close the bug for now.

Thanks

closed 2019-09-20 by icarus31