nordicsemi/IOS-nRF-Mesh-Library #158: Add acknowledge vendor message

icarus31, 2019-09-22:
**Is your feature request related to a problem? Please describe.**
Not really, mostly an enhancement based on the latest commit. I think the vendor message should also has the acknowledge message version.

**Describe the solution you'd like**
```
public protocol VendorMessage: MeshMessage {
    // No additional fields.
}

public protocol AcknowledgedVendorMessage: GenericMessage, StaticAcknowledgedMeshMessage {
    // No additional fields.
}
```

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

philips77, 2019-09-22:
Sure. Also, i think I'll remove the 4th delegate method, that i added on Friday. The other fail callback will be called, after all, if a status message isn't received, the node should assume it wasn't received.

icarus31, 2019-09-22:
I noticed something with vendor model when doing what I suggested above.

The first time I go to my view controller and I send a Get or Set message, the Status message is received, but the responseType is not executed.

```
public struct MyModeSet: AcknowledgedVendorMessage {
    public static var opCode: UInt32 = myModelOpCode.rawValue
    public static let responseType: StaticMeshMessage.Type = MyModelStatus.self
```
If I leave the view and come back, it is then executed.

Not sure it is clear enough. Just le me know. :-)

icarus31, 2019-09-22:
Do you think

```
public protocol VendorMessage: MeshMessage {
    // No additional fields.
}
```

should be changed for

```
public protocol VendorMessage: StaticMeshMessage {
    // No additional fields.
}
```

philips77, 2019-09-24:
I've added `AcknowledgedVendorMessage` and `AcknowledgedStaticVendorMessage`.
In your case you should extend the latter one.
Regarding your last question, there is `VendorMessage` and `StaticVendorMessage`. The difference is that in Static versions the response code is known during compilation.
In here: https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/version2/Example/nRFMeshProvision/View%20Controllers/Network/Configuration/Model/VendorModelViewCell.swift I'm using a `VendorMessage` type, as the opCode isn't known on compile time. Also, because there's no way to provide the response op code for Acknowledged message in this view, the RuntimeVendorMessage is just partially working, as it's sent as Unacknowledged message even if the switch is on. To change/fix this, an additional view for the response op code would be required, but this has super low priority.