nordicsemi/IOS-nRF-Mesh-Library #160: Unbinding app key make the status alert stays on forever

icarus31, 2019-09-23:
**Describe the bug**
When unbinding the app key from a model, the 'Status alert' stays on even if the action succeeded. I have similar situation when receiving a model status message.

If you tap 'cancel', the app key is still showing, but going out from the view and coming back does not show the app key.

If you want to avoid the issue here is the steps once the node is provisioned

- Go to the model view, example Health Server
- Bind the app key
- Go back to previous view
- Go to the model view again
- Unbind the app key

=> No status alert staying there.

**To Reproduce**
Steps to reproduce the behavior:
1. Provision a node
2. Select any model (beside the configuration)
3. Bind an app key
4. Unbind the app key

**Expected behavior**
The status alert should disappear.

**Platform details:**
 - Device: iPhone 7
 - OS: iOS13
 - Library Version (or Commit): 9b394e15

**Logs / Screenshots**
Add any logs or screenshots that can help us debug the issue faster
[image]

Full log once 'unbind' is executed

```
2019-09-23 19:02:06.227759-0400 nRF Mesh-v2[832:273868] [FoundationModel] Sending ConfigModelAppUnbind(applicationKeyIndex: 0, elementAddress: 2, modelIdentifier: 2, companyIdentifier: nil) to: 0002
2019-09-23 19:02:06.228655-0400 nRF Mesh-v2[832:273868] [Access] Sending Access PDU (opcode: 0x803F, parameters: 0x020000000200)
2019-09-23 19:02:06.230090-0400 nRF Mesh-v2[832:273868] [UpperTransport] Sending Upper Transport PDU (encrypted data: 0x00CEE9C14A3F7CFA, transMic: 0x6BE600B5) encrypted using key: Unknown device's Device Key
2019-09-23 19:02:06.231460-0400 nRF Mesh-v2[832:273868] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 49, segO: 0, segN: 0, data: 0x00CEE9C14A3F7CFA6BE600B5)
2019-09-23 19:02:06.233352-0400 nRF Mesh-v2[832:273868] [Network] Sending Network PDU (ivi: 0, nid: 0x2E, ctl: 0, ttl: 5, seq: 49, src: 0001, dst: 0002, transportPdu: 0x23C8AFF7366B7744A975EDEA8141A07B, netMic: 0xD3616145) encrypted using Primary Network Key (index: 0)
2019-09-23 19:02:06.234376-0400 nRF Mesh-v2[832:273868] [Bearer] -> 0x002E6C0BBDFFC380B85D23C8AFF7366B7744A975EDEA8141A07BD3616145
2019-09-23 19:02:06.289033-0400 nRF Mesh-v2[832:272851] [Bearer] <- 0x002E415F3EFAC416B26D7E74130AAB6ECA94212FB4DEF61B1C
2019-09-23 19:02:06.289592-0400 nRF Mesh-v2[832:273868] [Network] Network PDU (ivi: 0, nid: 0x2E, ctl: 1, ttl: 5, seq: 52, src: 0002, dst: 0001, transportPdu: 0x7E74130AAB6ECA, netMic: 0x94212FB4DEF61B1C) received
2019-09-23 19:02:06.290655-0400 nRF Mesh-v2[832:273868] [LowerTransport] ACK (seqZero: 49, blockAck: 0x00000001) receieved (decrypted using key: Primary Network Key (index: 0))
2019-09-23 19:02:06.318724-0400 nRF Mesh-v2[832:272851] [Bearer] <- 0x002E37F2FEB12DA815F57683F1E485FB9951F4F8A70E2955FEFF7FF74771
2019-09-23 19:02:06.319740-0400 nRF Mesh-v2[832:274514] [Network] Network PDU (ivi: 0, nid: 0x2E, ctl: 0, ttl: 5, seq: 53, src: 0002, dst: 0001, transportPdu: 0x7683F1E485FB9951F4F8A70E2955FEFF, netMic: 0x7FF74771) received
2019-09-23 19:02:06.321599-0400 nRF Mesh-v2[832:274514] [LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 53, segO: 0, segN: 1, data: 0x15B39E65B229BE1437275897) receieved (decrypted using key: Primary Network Key (index: 0))
2019-09-23 19:02:06.349711-0400 nRF Mesh-v2[832:272851] [Bearer] <- 0x002E2E6142CA77A46FA75DDD016EFDF8861B5D
2019-09-23 19:02:06.350758-0400 nRF Mesh-v2[832:274514] [Network] Network PDU (ivi: 0, nid: 0x2E, ctl: 0, ttl: 5, seq: 54, src: 0002, dst: 0001, transportPdu: 0x5DDD016EFD, netMic: 0xF8861B5D) received
2019-09-23 19:02:06.352111-0400 nRF Mesh-v2[832:274514] [LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 53, segO: 1, segN: 1, data: 0xC0) receieved (decrypted using key: Primary Network Key (index: 0))
2019-09-23 19:02:06.352321-0400 nRF Mesh-v2[832:274514] [LowerTransport] Access Message (akf: 0, szmic: 0, data: 0x15B39E65B229BE1437275897C0) received
2019-09-23 19:02:06.352698-0400 nRF Mesh-v2[832:274514] [LowerTransport] Sending ACK (seqZero: 53, blockAck: 0x00000003)
2019-09-23 19:02:06.353824-0400 nRF Mesh-v2[832:274514] [Network] Sending Network PDU (ivi: 0, nid: 0x2E, ctl: 1, ttl: 5, seq: 50, src: 0001, dst: 0002, transportPdu: 0xCB55B8892B9D28, netMic: 0x6AE6CD1E5CDF30D0) encrypted using Primary Network Key (index: 0)
2019-09-23 19:02:06.354137-0400 nRF Mesh-v2[832:274514] [Bearer] -> 0x002E5EC54059803AA3BFCB55B8892B9D286AE6CD1E5CDF30D0
2019-09-23 19:02:06.355196-0400 nRF Mesh-v2[832:274514] [UpperTransport] Upper Transport PDU (encrypted data: 0x15B39E65B229BE1437, transMic: 0x275897C0) received
Invalidating context
2019-09-23 19:02:06.355516-0400 nRF Mesh-v2[832:274514] [Access] Response Access PDU (opcode: 0x803E, parameters: 0x00020000000200) receieved (decrypted using key: Unknown device's Device Key)
2019-09-23 19:02:06.355848-0400 nRF Mesh-v2[832:274514] [FoundationModel] ConfigModelAppStatus(applicationKeyIndex: 0, elementAddress: 2, modelIdentifier: 2, companyIdentifier: nil, status: Success) received from: 0002
```

philips77, 2019-09-24:
This should work now. Another breaking feature from iOS 13.

closed 2019-09-24 by philips77

icarus31, 2019-09-24:
Wow you are fast! It also solved at the same time another issue that I had. Great work!