nordicsemi/IOS-nRF-Mesh-Library #161: Multiple MeshNetworks and sync

fiveagle, 2019-09-24:
Hi,
My app requirements are:
1- To be able to provision x nodes in a determined location and then do the same thing in a different location using potentially the same addresses, i.e. use different mesh network based on where I am.
2- It also needs to be able to recreate a working mesh network created by a different device (potentially an android device) downloading it from an API service. 

I need to understand if there is any feature already available in the sdk which can be used to manage this user scenario (Point 1). (Could it be to create multiple networks?)
A big problem for the point 2 is that the android sdk doesn't use the json codable approach. It saves much more things in a db. Which information are mandatory for recreating a working mesh network?

Thank You very much.

philips77, 2019-09-24:
Both iOS and Android mesh libraries can export and import compatiblity json configuration files. However, each device need to be set as a different provisioner with a different address. You may also have 2 managers with different networks at the same time. You may switch between them in one manager as well. The seq numbers are saved in user defaults per mesh network.
Regarding Android lib, looks like it's similar to https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/issues/251

fiveagle, 2019-09-25:
Does it mean that I cannot use the same Provisioner on different devices? Do I need to add a provisioner for each new device is using the json synced across the Cloud Api? Does this approach can potentially influence any operation on the mech devices? Like net key and app key binding or model subscriptions?

Thanks

philips77, 2019-09-25:
Yes, each device needs to be a different provisioner and have a different Unicast Address. Nodes keep track of the last SEQ number received form each Unicast Address and require it to be always greater than the last one. If you'd use the same Provisioner on multiple phones, they would start counting this SEQ numbed on their own and only the "fastest" one (the one which SEQ number is the highest) could be able to send messages.

Ideally, you need to set up all keys using one device, define other provisioners to be used on other phones and assign them ranges and provision as many devices as possible, and only then export the configuration to other nodes. Each other device, after importing the database configuration, should set a unique provisioner as local one and only then use the mesh. They can also provision their own nodes using the addresses from defined ranges (to avoid address collisions) or create groups, also from the range defined. When you start adding new groups/keys/nodes to the network, you may wish to merge the configurations at some point, so that all phones have the same network settings. Currently, the iOS and Android apps just replace databases, but we plan to add merge feature as well if the imported and current mesh network UUID are the same.

closed 2019-10-29 by philips77