nordicsemi/IOS-nRF-Mesh-Library #162: What feature was finnished?

Baothien1301, 2019-09-25:
Hi. 
I want to use app to control SIG mesh lights (HSL, CTL, lightness, vendor, scense, level, on off...).
For now, i modify app nrf mesh at branch master and control OK, i also modify provision process --> it does not fail in get composition step.
But i still look forward to your app, because i just learn and code app ios about 3 mounth and need learn more.

philips77, 2019-09-25:
So far, only Generic OnOff, Generic Level and Vendor messages are implemented from your list. I'll see if I can add lightning messages as well, but only to the library, not the sample app.
Also, use branch *version2*, not *master*. I'll be merging version 2 into master in few days.

philips77, 2019-10-29:
Hi, I'm about to publish the version 2 on master. Just FYI.
The other models will have to go to upcoming versions.
If you have them implemented already, you may do a PR so we can quickly add them :) Appreciated!

closed 2019-10-29 by philips77