nordicsemi/IOS-nRF-Mesh-Library #163: Response to Access PDU not received (timeout)

icarus31, 2019-09-25:
**Describe the bug**
I am getting few nodes where the Access PDU is not received when requesting composition data. I just wanted to know if this issue has been seen at your end and understand what is going on.

**To Reproduce**
Steps to reproduce the behavior:
1. Already have 20ish nodes provisioned very close to other 40 not provisioned (not sure if this has an impact, but wanted to give you my setup)
2. Provision the node
3. Start requesting the composition data
4. See error (timeout)

If I start empty mesh (no nodes provisioned) and try to get composition data for those 'weird' nodes' it goes very fast without any issue.

**Expected behavior**
Because I don't fully understand what is going on, not sure what the application should do. Does retrying again will help. Not that sure.

**Platform details:**
 - Device: iPhone7
 - OS: iOS13
 - Library Version (or Commit): 3e616836d6131

**Logs / Screenshots**
```
2019-09-25 13:37:27.204158-0400 nRF Mesh-v2[1215:529867] [Provisioning] Provisioning started
2019-09-25 13:37:27.204900-0400 nRF Mesh-v2[1215:529867] [Provisioning] Sending Provisioning Start (algorithm: FIPS P-256 Elliptic Curve, public Key: No OOB Public Key, authentication Method: No OOB)
2019-09-25 13:37:27.205316-0400 nRF Mesh-v2[1215:529867] [Bearer] -> 0x03020000000000
2019-09-25 13:37:27.206674-0400 nRF Mesh-v2[1215:529867] [Provisioning] Sending Provisioner Public Key (0xD7770D6C7783AD4FC77EFB1C4D8A0A4CF67193A93D23B6B13FF76EF3DFEA8BE572F35199D7C7F481FE245E336FDD8BC7E516F7D083B2F2AFEC413D417F77B43C)
2019-09-25 13:37:27.207620-0400 nRF Mesh-v2[1215:529867] [Bearer] -> 0x0303D7770D6C7783AD4FC77EFB1C4D8A0A4CF67193A93D23B6B13FF76EF3DFEA8BE572F35199D7C7F481FE245E336FDD8BC7E516F7D083B2F2AFEC413D417F77B43C
2019-09-25 13:37:27.419172-0400 nRF Mesh-v2[1215:529867] [Bearer] <- 0x0303666792BA418B65D3207F51F17BF0E9737250478BD2B8AEB6465C677E675A07725C20BFE93D2B1E7D0592F3EFE31837C7BCAE178DF605ED3D982BB12072890F55
2019-09-25 13:37:27.421955-0400 nRF Mesh-v2[1215:529867] [Provisioning] Device Public Key (0x666792BA418B65D3207F51F17BF0E9737250478BD2B8AEB6465C677E675A07725C20BFE93D2B1E7D0592F3EFE31837C7BCAE178DF605ED3D982BB12072890F55) received
2019-09-25 13:37:27.428812-0400 nRF Mesh-v2[1215:529867] [Provisioning] Sending Provisioner Confirmation (0x5B3E6510F24B4E36A9D774F6EA0FB946)
2019-09-25 13:37:27.429534-0400 nRF Mesh-v2[1215:529867] [Bearer] -> 0x03055B3E6510F24B4E36A9D774F6EA0FB946
2019-09-25 13:37:27.476287-0400 nRF Mesh-v2[1215:529867] [Bearer] <- 0x0305FFD3C8EEF189662F7E1C131F4FF3984A
2019-09-25 13:37:27.477279-0400 nRF Mesh-v2[1215:529867] [Provisioning] Device Confirmation (0xFFD3C8EEF189662F7E1C131F4FF3984A) received
2019-09-25 13:37:27.478268-0400 nRF Mesh-v2[1215:529867] [Provisioning] Sending Provisioner Random (0x2305F0CB8C134327A9005D00F570F3CE)
2019-09-25 13:37:27.479667-0400 nRF Mesh-v2[1215:529867] [Bearer] -> 0x03062305F0CB8C134327A9005D00F570F3CE
2019-09-25 13:37:27.536202-0400 nRF Mesh-v2[1215:529867] [Bearer] <- 0x030696E7F15100D851493C8DF9A08BA6BC7F
2019-09-25 13:37:27.537067-0400 nRF Mesh-v2[1215:529867] [Provisioning] Device Random (0x96E7F15100D851493C8DF9A08BA6BC7F) received
2019-09-25 13:37:27.539706-0400 nRF Mesh-v2[1215:529867] [Provisioning] Sending Encrypted Provisioning Data (0x9EF580CC4E2C9A27F21F94B472B82E8DAF3F66D5CFFE9047DF70DABD6593D9E5A9)
2019-09-25 13:37:27.541615-0400 nRF Mesh-v2[1215:529867] [Bearer] -> 0x03079EF580CC4E2C9A27F21F94B472B82E8DAF3F66D5CFFE9047DF70DABD6593D9E5A9
2019-09-25 13:37:27.595005-0400 nRF Mesh-v2[1215:529867] [Bearer] <- 0x0308
2019-09-25 13:37:27.595947-0400 nRF Mesh-v2[1215:529867] [Provisioning] Provisioning Complete received
2019-09-25 13:37:27.597146-0400 nRF Mesh-v2[1215:529867] [Provisioning] Provisioning complete
2019-09-25 13:37:31.251499-0400 nRF Mesh-v2[1215:529867] [Bearer] Cancelling connection...
2019-09-25 13:37:31.263309-0400 nRF Mesh-v2[1215:529867] [Bearer] Disconnected from node e5:c8
2019-09-25 13:37:31.264609-0400 nRF Mesh-v2[1215:529867] [Provisioning] Provisioner is ready
2019-09-25 13:37:31.836579-0400 nRF Mesh-v2[1215:530072] [CoreBluetooth] XPC connection invalid
2019-09-25 13:37:31.843596-0400 nRF Mesh-v2[1215:530072] [CoreBluetooth] XPC connection invalid
2019-09-25 13:37:31.852105-0400 nRF Mesh-v2[1215:529867] [TableView] Warning once only: UITableView was told to layout its visible cells and other contents without being in the view hierarchy (the table view or one of its superviews has not been added to a window). This may cause bugs by forcing views inside the table view to load and perform layout without accurate information (e.g. table view bounds, trait collection, layout margins, safe area insets, etc), and will also cause unnecessary performance overhead due to extra layout passes. Make a symbolic breakpoint at UITableViewAlertForLayoutOutsideViewHierarchy to catch this in the debugger and see what caused this to occur, so you can avoid this action altogether if possible, or defer it until the table view has been added to a window. Table view: <UITableView: 0x10290a200; frame = (0 116; 375 551); clipsToBounds = YES; autoresize = W+H; tintColor = UIExtendedSRGBColorSpace -0.069266 0.548305 0.825235 1; gestureRecognizers = <NSArray: 0x282ef9380>; layer = <CALayer: 0x28203fd80>; contentOffset: {0, 0}; contentSize: {0, 0}; adjustedContentInset: {0, 0, 0, 0}; dataSource: <nRF_Mesh_v2.ConfigurationViewController: 0x101a5e660>>
2019-09-25 13:37:32.403985-0400 nRF Mesh-v2[1215:530072] [FoundationModel] Sending ConfigCompositionDataGet(page: 0) to: 0005
2019-09-25 13:37:32.405538-0400 nRF Mesh-v2[1215:530072] [Access] Sending Access PDU (opcode: 0x8008, parameters: 0x00)
2019-09-25 13:37:32.408708-0400 nRF Mesh-v2[1215:530072] [UpperTransport] Sending Upper Transport PDU (encrypted data: 0xE7E39C, transMic: 0xDB10BC08) encrypted using key: S50 A10's Device Key
2019-09-25 13:37:32.411168-0400 nRF Mesh-v2[1215:530072] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 23, segO: 0, segN: 0, data: 0xE7E39CDB10BC08)
2019-09-25 13:37:32.421243-0400 nRF Mesh-v2[1215:530072] [Network] Sending Network PDU (ivi: 0, nid: 0x0B, ctl: 0, ttl: 5, seq: 23, src: 0001, dst: 0005, transportPdu: 0x0A621AA822B20619C5534F, netMic: 0xCACEF748) encrypted using Primary Network Key (index: 0)
2019-09-25 13:37:32.422772-0400 nRF Mesh-v2[1215:530072] [Bearer] -> 0x000BAF61B148252A6DBF0A621AA822B20619C5534FCACEF748
2019-09-25 13:37:32.536848-0400 nRF Mesh-v2[1215:529867] [Bearer] <- 0x000B35F329D487BEC1270D3EF8AD29C8D75E31200B2434A85E174A68B274
2019-09-25 13:37:32.537538-0400 nRF Mesh-v2[1215:530074] [Network] Network PDU (ivi: 0, nid: 0x0B, ctl: 0, ttl: 4, seq: 1, src: 0005, dst: 0001, transportPdu: 0x0D3EF8AD29C8D75E31200B2434A85E17, netMic: 0x4A68B274) received
2019-09-25 13:37:32.539489-0400 nRF Mesh-v2[1215:530074] [LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 1, segO: 0, segN: 6, data: 0xF8DB302212C8AB852473EC48) receieved (decrypted using key: Primary Network Key (index: 0))
2019-09-25 13:37:32.567859-0400 nRF Mesh-v2[1215:529867] [Bearer] <- 0x000B1149F4071BE548CB0E4479C7D0DF010B279DED6DFE74A57281B87861
2019-09-25 13:37:32.569412-0400 nRF Mesh-v2[1215:530074] [Network] Network PDU (ivi: 0, nid: 0x0B, ctl: 0, ttl: 4, seq: 6, src: 0005, dst: 0001, transportPdu: 0x0E4479C7D0DF010B279DED6DFE74A572, netMic: 0x81B87861) received
2019-09-25 13:37:32.571395-0400 nRF Mesh-v2[1215:530074] [LowerTransport] Segmented Access Message (akf: 0, szmic: 0, seqZero: 1, segO: 5, segN: 6, data: 0x2CECDEAB809C237CE5F7CF34) receieved (decrypted using key: Primary Network Key (index: 0))
2019-09-25 13:37:32.597417-0400 nRF Mesh-v2[1215:529867] [Bearer] <- 0x000B29FA59490EFFE07465393292618BEEAC93C8BCEC5375A1844B092CCC
2019-09-25 13:37:32.599196-0400 nRF Mesh-v2[1215:530074] [Network] Network PDU (ivi: 0, nid: 0x0B, ctl: 0, ttl: 3, seq: 5, src: 0005, dst: 0001, transportPdu: 0x65393292618BEEAC93C8BCEC5375A184, netMic: 0x4B092CCC) received
2019-09-25 13:37:32.599870-0400 nRF Mesh-v2[1215:530074] [LowerTransport] Discarding packet (seqAuth: 5, expected > 6)
2019-09-25 13:37:32.627865-0400 nRF Mesh-v2[1215:529867] [Bearer] <- 0x000B6146B5CDD42A566A6AF1EB0AFCF15084DD87FA54177089EB011DB128
2019-09-25 13:37:32.629846-0400 nRF Mesh-v2[1215:530074] [Network] Network PDU (ivi: 0, nid: 0x0B, ctl: 0, ttl: 3, seq: 4, src: 0005, dst: 0001, transportPdu: 0x6AF1EB0AFCF15084DD87FA54177089EB, netMic: 0x011DB128) received
[... cut ...]
2019-09-25 13:37:50.709413-0400 nRF Mesh-v2[1215:530074] [Bearer] -> 0x000BF4FEEA34434113F3DB8F72A0482C303EE87F10B70C425F
2019-09-25 13:37:59.831314-0400 nRF Mesh-v2[1215:530072] [LowerTransport] Incomplete message timeout: cancelling message (src: 0005, seqZero: 68)
2019-09-25 13:38:02.439146-0400 nRF Mesh-v2[1215:530072] [Access] Response to Access PDU (opcode: 0x8008, parameters: 0x00) not received (timeout)
2019-09-25 13:38:02.441142-0400 nRF Mesh-v2[1215:530072] [FoundationModel] ConfigCompositionDataGet(page: 0) sent from: 0001, to: 0005 timed out

```

philips77, 2019-09-25:
I see a lot of packets being discarded as they came in a wrong order. I'll look into it tomorrow.

icarus31, 2019-09-25:
Yes, you are right about discarded packets. 

I also forgot to mention the nodes are really close together and have a -20db attenuator. So, we are transmitting at -10dbm approx.

I did other tests and by disconnecting the power for lots of nodes, it helped to get the "bad" nodes completely provisioned and configured.

philips77, 2019-09-27:
Could you try with the latest commits?
What has changed:
1. The acknowledged message is repeated after 5 sec, not 2. This gives more time to complete the first attempt.
2. The SEQ validation for segmented messages is done against SeqZero of the first segment. Segments may be received in any order and they will be accepted.
3. I made the segmentation thread safe.

icarus31, 2019-09-27:
Hi,

Just tried you latest release and the provisioning went much more smoothly. I still got the same errors for few nodes (5 over 50) and even with the retry, it did not resolve by itself. Doing a refresh to get the composition data did the work.

Only one did not want to finish the configuration. I will need to investigate it.

I asked the test team to provision more nodes, so I will see how it goes and give you an update.

philips77, 2019-09-27:
I pushed some more updates, e.g. fixed a deadlock when sending a message to self.
After all those changes, the Provisioner will retry sending Acknowledged Messages until the response is received, or it times out. Responses are processed on a concurrent queue, but reassembly is done using a serial queue as race conditions were possible. The replay protection check is done against `SeqAuth`, which for segmented messages is calculated using seqZero, not the packet's sequence number, so segments may are accepted in any order.
Besides, I fixed issues when the Blacklist proxy filter was set. The Node was replying to messages targeting other Nodes, which was a bug.

Lots of bugs fixed today.

icarus31, 2019-09-27:
Good work!
I will give it a try and keep you updated! Thanks

icarus31, 2019-09-30:
The feedback that I got from the test people is the provisioning of our test nodes went good to be able to complete their task.

They still got ones that did not want to get the composition data, but once they reset the node, the next try went smoothly.

Also, having a proxy node closer to the unprovisioned nodes helped to get it through.

I am supposed to have a meeting with them today to get more details. I will let you know if I get something new.

philips77, 2019-09-30:
Thanks for the update. 
FYI: I'm refactoring the access/foundation layers at the moment. Expect a PR today/tomorrow. It will break many things, but hopefully just a little bit.
To handle messages, instead of relying on what's implemented in `AccessLayer` (this huuuuge switch), or registering vendor models in the manager, there will be model handlers, which can handle messages per model. The handles will have to be set during setting up the local elements.

philips77, 2019-10-09:
FYI: The PR has been merged.
Now, to handle messages (Generic or Vendor) you have to add the model to the local elements. In the model you have to handle the incoming messages. Follow the `AppDelegate` in the sample app for an example.

closed 2019-10-21 by philips77

PrakrutiVolansys, 2022-04-05:
@philips77 can you please let me know in which release, the fix was released?