nordicsemi/IOS-nRF-Mesh-Library #164: Can't not found elements

codepgq, 2019-09-29:
I download latest code, i can provision device, but can't found elements.  if i use android phone, is work. Add Application Keys is not work.

[image]

[image]

'
[image]

philips77, 2019-09-29:
Looks like your provisioner has disabled configuration (doesn't have unicast address assigned), thus it can't send messages to the network. 
Go to Settings, than Provisioners, click the local Provisioner and assign any available address from it's range.

codepgq, 2019-09-29:
thanks，is work now

closed 2019-09-29 by codepgq