nordicsemi/IOS-nRF-Mesh-Library #165: Add a proxy node selector

icarus31, 2019-09-30:
**Is your feature request related to a problem? Please describe.**
By default, the application is selecting a proxy node. This is very nice compare to version one, but some time, it would be nice to select a specific one from nearest nodes.

**Describe the solution you'd like**
When selecting a node, the application will try to connect to the proxy selected.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

philips77, 2019-09-30:
Yeah... I understand. Some users want it to be automatic, some just want to connect to the network. For me, the proxy connection should be transparent to the user, as after all, it's mesh. When Apple one day decides to open Mesh Advertising Ids, we could switch the bearer to ADV.
I also sometimes find the need to connect to specific node, but I just disconnect all other nodes (I don't have many of them). You may use the Android sample app for nRF Mesh, which allows to select a proxy, if you need to test it.
Also, if you'd check `NetworkConnection` class, it allows to connect to multiple proxies at the same time (by default only 1), and if a connection is dropped, seamlessly switch to the next one.

philips77, 2019-11-04:
Please, try the version on the PR's branch.

philips77, 2019-11-06:
Released in version 2.1.0.

closed 2019-11-06 by philips77